klausahrenberg/WThermostatBeca #1: AC with Fan speed

avariorss, 2019-03-13:
Hi, Nice work, thanks for sharing!!

I have a slightly different model (looks identical from the front and same PCB): BAC-002ALW. This one is for cooling and has a 3 speed fan control.

I'd love to help get your platform working on a wider range of these devices: I'd be very happy to hook up an FTDI and read off the hex codes it spits out from the ESP chip... Not sure how much help I can be on the code side of things, but happy to help if I can. 

[image]

ahmaddxb, 2019-04-12:
I also have the 3 speed fan version is there any control for the fan speed?

ahmaddxb, 2019-04-12:
@avariorss Did you flash this firmware on your thermostat and test it? How did it function? I have the exact same model as you and are looking for a way to integrate it with openhab over mqtt.

klausahrenberg, 2019-04-13:
Hi, I think to support this additional fan wouldn't be a problem - but I don't have this device and so I don't know, how the serial commands between the ESP and device controller are.
Try to flash this firmware, BUT make an backup of the original before - so you can go back to your original device every time! After flashing and configuring  of your MQTT broker try to set the fan speed on the device. The firmware should send now an error message with "command unknown" to your broker. Try this for every function on your device and send me the related unknown-Message. Maybe it's not a big think to integrate also your device. BUT: make a backup of the original firmware before, I'm not responsible for any damage on your device - backup procedure is also a good test for flashing later, if backup is not working, flashing will also not be possible.

ahmaddxb, 2019-04-20:
Can you tell me pins i need to tap into to get RX TX 3.3V and Ground from the board. Also which pin do i need to short while powering up the board to get it into programming mode.

Also if i want to make a backup of original firmware does the board need to be booted in programming mode or just powered on normally. Thanks

I have ordered a second unit to do tests on which will be coming shortly from Aliexpress.

klausahrenberg, 2019-04-20:
Here is a description of the pinout of the Tywe-Module, it's nearly the same as an ESP-12F module: https://github.com/arendst/Sonoff-Tasmota/wiki/TYWE3S
Connect RX, TX, GND and 3.3V. Gpio0 must be connected to GND during Power up to get it in programming mode. For backup you also need the module in programming mode first. After backup you have to bring the module in programming mode again before flashing.

ahmaddxb, 2019-04-20:
Thanks for you reply. I'll let you know how i get on when i receive the unit.

ahmaddxb, 2019-04-29:
Hi i got the device today but I'm unable to flash it. I have grounded Gpio0 and used the command below.

C:\esptool-2.6>esptool.py -p COM4 -b 460800 read_flash 0x00000 0x100000 originalFirmware1M.bin
esptool.py v2.6
Serial port COM4
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header

C:\esptool-2.6>

klausahrenberg, 2019-04-29:
Timeout means, connection to chip not possible. Some possible reasons, please check:
- VCC must be 3.3V, not 5V. GPIO0 must connected to GND from start of supply voltage on - not afterwards (Both points should be clear, just written again)
- COM 4 is right? If everything is well connected, you should see the COM4 in your Arduino-IDE as an available Port
- Which programmer did you used? I had some problems with CH340 programmer, I prefer to use a CP2101 programmer
- Maybe reducing the baud rate also helps, try flag "-b 57600" instead of "-b 460800"
- RX and TX connection must be crossed: RX at ESP must be connected to TX at programmer and TX-ESP to Programmer-RX
- Connect also the CH_PD (also known as EN) pin on the ESP to 3.3V for programming
If points above does not help:
- Try to use an Arduino UNO as programmer, it's slow, but solid. Here is the instruction: https://www.youtube.com/watch?v=ij_jlEHufYQ In this case, don't cross RX and TX
Good luck

ahmaddxb, 2019-04-29:
I am using the Arduino UNO as the programmer. when i open serial monitor in Arduino-IDE I see an output so i know my tx and tx pins are correct. I have just tried to do a backup on a sonoff basic and that has also failed so im going to try to successfully backup the sonoff and when i get it working move on the the beca.

klausahrenberg, 2019-04-29:
Ok, but with an UNO you should reduce the baud rate, I think

klausahrenberg, 2019-04-29:
Or remove the baud rate flag and let esp tool figure out

ahmaddxb, 2019-04-29:
Manage to get the sonoff working on another computer and was able to do backups but im not able to do the same with the beca. Even trying the command esptool.py --port COM3 flash_id fails to communicate. I have now plugging into the mains to tested if its working correctly and have paired it with the app and it all function fine. I don't know why it hasn't worked for me it seams to be the exact same board.

[image]

a-roz, 2019-04-29:
How do you power PCB ? Have to be powered only TYWE module. If sonix CPU powered as well, it disturbs communication on tx/rx lines, and the module doesn't enter to programming mode. 
I had the same problem, and after few investigation was solved. Or cardinally you can cut 2 pcb routes tx and rx, which connects module to CPU and solder if after programming finished.

ahmaddxb, 2019-04-29:
Did you remove the esp module from the main pcb. What i did is exactly how it looks in the image i just solder some wires so i can connect them to my Arduino Uno. If i remove the connection from the tx and rx to the main pcb will this solve my issue you are saying?

a-roz, 2019-04-29:
Noooo, I did not remove module. 
Wires was soldered as you did.
I did not used Uno like a programmer, have use FTDI232-TTL-3.3 serial converter. 
Also use external 3.3 power supply to feed only module and connected to your BROWN.
On the YELLOW wire I've install button to GND which used to enter prog.mode.

The only problem I can see probably you have solder wires to close to metal cover of module with possibility of making shortcut.

klausahrenberg, 2019-04-29:
Your wiring looks correct, but the hint from a-roz is correct, maybe you have a shortcut between RX and TX through the metal housing. I had this issue too in the past and was looking hours until I detected a little bridge between the wires. Additionally I can only repeat, try to solder the EN/CH_PD additionally and connect this to VCC. Another thing is, but i can't imagine an influence here: you removed the flat wire between control unit and relay unit - as I remember, I left this connected during flashing. However, don't connect the main AC to the unit during flashing.

ahmaddxb, 2019-04-29:
ok i managed to do it, here is what i needed to do.

1.) connected Arduino as my vcc, ground and gpio0 to the esp but didnt connect the usb to the computer yet.

2. connect another FTDI FT232RL USB to TTL Serial Adapter to the tx and rx pin (connected to the computer usb)

[image]

3. run the command esptool.py -p COM4 -b 460800 read_flash 0x00000 0x100000 originalFirmware1M.bin

4. plug in the Arduino usb to my computer and pull out the Gpio0 connector.

Finally got my backup

ahmaddxb, 2019-04-29:
I flashed your firmware now and connected it the the wifi and i have its dhcp ip address. should i be able to access the web gui from the ip in the browser?

ahmaddxb, 2019-04-29:
This are the commands i get from switching the fan settings.
Fan speed

Auto - 		{"unknown":"55 aa 01 07 00 05 67 04 00 01 00"}
Speed 1 - 	{"unknown":"55 aa 01 07 00 05 67 04 00 01 02"}
speed 2 -	{"unknown":"55 aa 01 07 00 05 67 04 00 01 02"}
Speed 3 -	{"unknown":"55 aa 01 07 00 05 67 04 00 01 01"}

I can also switch the mode in my model from cooling , heating and ventilation but i get no command back from that?

klausahrenberg, 2019-04-30:
Hi, good to know that flashing worked now. To your question 
>should i be able to access the web gui from the ip in the browser? 
 
No, the webServer stays disabled, if the Wifi was configured. The Web-Gui has no information about the thermostat, all is send to MQTT. You need the Web-Gui only for firmware update. To enable the Web-Gui for flashing you have 2 posibilities: 1. Switch the thermostat off and hold the down-Button for 5 seconds, until the display is flashing - or 2. Send a "webServer"-command with payload "true" via MQTT (see README.md). After that you can access to the Web-GUI via browser to update the firmware. By the way: You could also flash the original firmware (your backup) here to get your device back to factory state.

But back to your fan speed: Your unknown commands look strange because speed 1 and speed 2 are identical commands. I implemented the commands above, please test.
Flash it via Web-GUI, after flash you should see firmware-version 0.8 and an additional info "fanSpeed" in the json. At beginning there is value "none", but it should change between "auto", "low", "medium" and "high". You can also send an MQTT command <YOUR_TOPIC>/fanSpeed; payload: "auto"|"low"|"medium"|"high". Try and have a look, if your thermostat switches between the fan speeds. Because of your identical commands, I believe that this will not work correctly from beginning, but it should switch something - but maybe wrong. Please report.

Your other problem
>I can also switch the mode in my model from cooling , heating and ventilation but i get no command back from that?

Not all states of the devices are reported to the ESP, e.g. at my model, there is no heating flag that indicates, if the heating is actual on. 
Was you able to switch these modes in the Tuya-Application before? If yes - there must be a way that I actual not know yet. If not, it will be not possible to do this via MQTT. 
Are you switching between these modes with the left button? At my device the left button switches between "schedule" and "manual" mode. Is the order at your device: "schedule", "manual", "cooling", "heating",... or similar? If yes, have a look in this test version, if there is a new unknown command send like "55 AA 00 07 00 05 04 04 00 01 0x"

[ThermostatBecaWifi_test.zip](https://github.com/klausahrenberg/ThermostatBecaWifi/files/3129562/ThermostatBecaWifi_test.zip)

ahmaddxb, 2019-04-30:
sorry typo in Fan speed 1

Auto - 		{"unknown":"55 aa 01 07 00 05 67 04 00 01 00"}
Speed 1 - 	{"unknown":"55 aa 01 07 00 05 67 04 00 01 03"}
speed 2 - 	{"unknown":"55 aa 01 07 00 05 67 04 00 01 02"}
Speed 3 - 	{"unknown":"55 aa 01 07 00 05 67 04 00 01 01"}

The left button switches between the cooling heating and ventilation modes on my unit. This was also possible through the app. In the manual it says that the left button is for switching from manual mode to programming mode but it doesn't do that on the unit. The only way i was able to do it was from the tuya app but with your firmware it works with the command manualMode/false or true so that's fine.

I tried your test firmware and the fanSpeed functions correctly. Thanks

When I first start up the unit i get this code with the test firmware.

{"unknown":"55 aa 01 07 00 3a 68 00 00 36 00 06 28 00 08 1e 1e 0b 1e 1e 0d 1e 00 11 2c 00 16 1e 00 06 28 00 08 28 1e 0b 28 1e 0d 28 00 11 28 00 16 1e 00 06 28 00 08 28 1e 0b 28 1e 0d 28 00 11 28 00 16 1e"}

version 0.7
I have found that the command locked works with the true payload but with the false payload the screen lights up but doest unlock the unit.

Command webService/true or false doesn't work for me.

If i change the set temperature on the unit should i be getting some feedback over mqtt because I don't.

command state/0 is reporting correct data over mqtt when i request for it.

ahmaddxb, 2019-04-30:
schedules are also not pulled off the device correctly, they all appear to be empty but i have set it up on the unit with values but they are not being read.

{"workday":{"0":{"h":"00:00","t":0},"1":{"h":"00:00","t":0},"2":{"h":"00:00","t":0},"3":{"h":"00:00","t":0},"4":{"h":"00:00","t":0},"5":{"h":"00:00","t":0}}}

{"saturday":{"0":{"h":"00:00","t":0},"1":{"h":"00:00","t":0},"2":{"h":"00:00","t":0},"3":{"h":"00:00","t":0},"4":{"h":"00:00","t":0},"5":{"h":"00:00","t":0}}}

{"sunday":{"0":{"h":"00:00","t":0},"1":{"h":"00:00","t":0},"2":{"h":"00:00","t":0},"3":{"h":"00:00","t":0},"4":{"h":"00:00","t":0},"5":{"h":"00:00","t":0}}}

ahmaddxb, 2019-04-30:
Your Example
<YOUR_TOPIC>/mcucommand sends directly serial mcu commands to the device. The has to be a string in hexa-form without the checksum buyte at the end, e.g. to set the desired temperature to 24.5C: "55 aa **01 07** 00 08 02 02 00 04 00 00 00 31"

For me your example doesn't work but if I use the code below it works. The characters in bold are differnet.

55 AA **00 06** 00 08 02 02 00 04 00 00 00 31

ahmaddxb, 2019-04-30:
I have notice also for my fan speed to work via the mcucommand reported back over mqtt dont work unless i change them see below difference in bold.

Reported back but dont work when issues by the mcucommand

Auto - 		{"unknown":"55 aa **01 07** 00 05 67 04 00 01 00"}
Speed 1 - 	{"unknown":"55 aa **01 07** 00 05 67 04 00 01 03"}
speed 2 - 	{"unknown":"55 aa **01 07** 00 05 67 04 00 01 02"}
Speed 3 - 	{"unknown":"55 aa **01 07** 00 05 67 04 00 01 01"}

mcucommand that work for me

Auto - 		{"unknown":"55 aa **00 06** 00 05 67 04 00 01 00"}
Speed 1 - 	{"unknown":"55 aa **00 06** 00 05 67 04 00 01 03"}
speed 2 - 	{"unknown":"55 aa **00 06** 00 05 67 04 00 01 02"}
Speed 3 - 	{"unknown":"55 aa **00 06** 00 05 67 04 00 01 01"}

klausahrenberg, 2019-04-30:
Yes, that's clear, **00 06** is a status report from device, if you want to set the values, you have to use **01 07**. I considered this already, it's for all commands the same. Good, that fan is working - we will bring this to a working solution, I think. Some questions to finalize the fan issue: 
1. Speed 1 is **03** and means "low speed", Speed 3 is **01** and means "high speed"? Please confirm. 
2. Are you able to set the fan speed via MQTT command with payload auto, low, medium, high?

ahmaddxb, 2019-04-30:
1. This is correct
2. Yes they all work via MQTT

klausahrenberg, 2019-04-30:
Regarding your not working webService. Are you using node-red to send the MQTT messages? If yes, insert these 3 little nodes, modify the correct topic and try to switch the webService on and off. This is working at my device.
[{"id":"e56f1c66.c7b88","type":"inject","z":"465b0d0a.e931d4","name":"on","topic":"","payload":"true","payloadType":"bool","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":70,"y":380,"wires":[["f18483dc.0a8c"]]},{"id":"f91599fb.7b0898","type":"inject","z":"465b0d0a.e931d4","name":"off","topic":"","payload":"false","payloadType":"bool","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":70,"y":420,"wires":[["f18483dc.0a8c"]]},{"id":"f18483dc.0a8c","type":"mqtt out","z":"465b0d0a.e931d4","name":"","topic":"lotte/livingroom/thermostat/webServer","qos":"","retain":"","broker":"65ce9e3c.59aa7","x":310,"y":380,"wires":[]},{"id":"65ce9e3c.59aa7","type":"mqtt-broker","z":"","name":"Lotte","broker":"localhost","port":"1883","clientid":"","usetls":false,"compatmode":true,"keepalive":"60","cleansession":true,"birthTopic":"","birthQos":"0","birthPayload":"","closeTopic":"","closeQos":"0","closePayload":"","willTopic":"","willQos":"0","willPayload":""}]

klausahrenberg, 2019-04-30:
And last your schedule issue:
I made a new test version :-) This version will forward almost all received mcu message to the MQTT broker. To solve your mode selection for example. Please push the left button and recognize, which command you receive - and send this to me.
Also there are a lot of messages at the start of device (including schedules) until the first "state" message come to your MQTT broker. Please send me all these raw messages.

[ThermostatBecaWifi_081_test.zip](https://github.com/klausahrenberg/ThermostatBecaWifi/files/3130384/ThermostatBecaWifi_081_test.zip)

ahmaddxb, 2019-04-30:
I'm not using node red right now I'm just using a public MQTT broker "broker.hivemq.com" and using the application MQTT.fx v1.7.1 on windows to see and send commands. I will take the device home an add it to my openhab broker after I am done testing.

ahmaddxb, 2019-04-30:
Startup messages

{"mcu: 0x01":"55 aa 01 07 00 05 01 01 00 01 01"}
{"mcu: 0x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 31"}
{"mcu: 0x04":"55 aa 01 07 00 05 04 04 00 01 01"}
{"mcu: 0x05":"55 aa 01 07 00 05 05 01 00 01 00"}
{"mcu: 0x06":"55 aa 01 07 00 05 06 01 00 01 00"}
{"mcu: 0x67":"55 aa 01 07 00 05 67 04 00 01 03"}

ahmaddxb, 2019-04-30:
When i press the left button it doesnt generate any mqtt messages

klausahrenberg, 2019-04-30:
Your startup messages are little different. At startup, the ESP query the state of the device with 1 command given by the Tuya-specification initial. And then my device sends more information than yours: the actual temperature (0x03), schedules (0x65) and floor temperature (0x66, external sensor, does your device have it?) are missing.
Please test one thing: Modify the schedules on the device (one time parameter is enough). Are you receiving now any command (0x65) ?

ahmaddxb, 2019-04-30:
I modified the scheduled on the device but there is no message in mqtt

klausahrenberg, 2019-04-30:
To make the things more easy: Please send me your backup of the original firmware. I will flash it to my device. Hopefully I can connect it to the Tuya app and receive the generated serial commands from the original firmware.

ahmaddxb, 2019-04-30:
[becaoriginalFirmware1M.zip](https://github.com/klausahrenberg/ThermostatBecaWifi/files/3130511/becaoriginalFirmware1M.zip)

ahmaddxb, 2019-04-30:
> Yes, that's clear, **00 06** is a status report from device, if you want to set the values, you have to use **01 07**. I considered this already, it's for all commands the same. Good, that fan is [...]

For me this is the opposite **00 06** is to set the value and **01 07** is a status report from device.

ahmaddxb, 2019-05-01:
How did you get on flashing my original firmware, did it work and could you connect to the beca app?

klausahrenberg, 2019-05-01:
I had not so much time so far. Yes, I flashed, but I'm struggling with connection to Tuya App. I tried different apps without success so far. Which one have you exactly used, a link would be welcome

ahmaddxb, 2019-05-01:
My BecaSmart - [Link](https://play.google.com/store/apps/details?id=com.beca.smart&hl=en)

klausahrenberg, 2019-05-01:
Thanks for the link. I used the right app. I flashed my original firmware, connecting was running well. Than I flashed again yours - no connection possible. I connected the RX and TX to track the communication and it seems, that the ESP sends a initialisation request for setting the time and my thermostat sonix controller answers not or not in the right way. So the connection try ends up in a endless loop at my device.

So it seems the following information for your thermostat I don't know and are not able to find out:
- commands for mode cooling or heating
- commands to get the schedules separatly
- commands for get the temperature/floor temperature (are you getting these values? because it was not send during initialization of the device, but normally send every 5 seconds)
- command to set the required temperature

To get these information you need a right device. At my device I have got all commands by connecting the RX and TX line (and GND) to a serial monitor and connect with 9600 baud. Then I was able to switch all commands in the Tyua app and so I received all the nice working commands on my serial monitor. This was my plan to do with your original firmware. However, because my device sonix controller didn't answer right, it was not possible.

klausahrenberg, 2019-05-01:
I just uploaded version 0.8. The fan support is still included.

> If i change the set temperature on the unit should i be getting some feedback over mqtt because I don't.

This issue should be fixed now. Reason see changes file. Please report, if you now receive an status update at MQTT, if you change the desired temperature

ahmaddxb, 2019-05-01:
This is looking good. Nice improvements

Working
Device turns on with a state update
Changes made on the device buttons like temp and fan cause state messages over mqtt

Not Working
LogMCU command is not working
webService command is not working
locked command with false payload still doesnt unlock the device however if i send mcucommand 55 AA 00 06 00 05 06 01 00 01 00 it will unlock correctly.

ahmaddxb, 2019-05-01:
Found the problem with webService command not working.

In your readme you have

<YOUR_TOPIC>/webService/true|false can be called with an bool to switch the web service on or off

It should be 

<YOUR_TOPIC>/webServer/true|false can be called with an bool to switch the web service on or off

However changing the webServer command doesnt invoke an automatic state update over mtqq

ahmaddxb, 2019-05-01:
> To get these information you need a right device. At my device I have got all commands by connecting the RX and TX line (and GND) to a serial monitor and connect with 9600 baud. Then I was able to [...]

What hardware do i need to collect these commands can i use my FTDI FT232RL USB to TTL Serial Adapter / arduino Uno and a bit of software? If that all i need let me know what software you used and to options that need to be set in the software to read the commands correctly.

klausahrenberg, 2019-05-01:
Version 0.9 fixes all your not working points - enough for today :-)
For logging the serial commands you can take your normal programmer equipment, just leave it connected after flashing. Following steps:
- After flashing disconnect the USB cable from your computer
- leave all connections at your programmer (in your case Arduino UNO). To be sure, you can remove the VCC cable because the ESP will be powered by the device. Also remove GPIO0 from GND - otherwise the ESP is in programming mode again.
- Connect the AC to your Thermostat
- Connect the USB cable at your computer again
- Open any serial monitor on your computer (e.g. Linux 'moserial', for Windows maybe 'pqcom'? I don't have Windows, but pqcom looks usable). Any serial monitor with a hex view is welcome.
- Connect with 9600 baud to serial port
You will see the output of the ESP module that is sent to the sonix device controller. The opposite way not. Your device will work normally. Everey second you should see a short heartbeat command. Now you can switch all commands in the Beca App und you will see the corresponding hex-commands in the monitor, every command starts with '55 AA ..'
Your help is welcome and I'll try to implement these functions, if you receive any output that makes sence. However, this is at your own risk :-)

ahmaddxb, 2019-05-01:
Amazing work, I really appreciate your help. I will try to do the serial commands at some point and report them back too you. 

If I see a command starting with "55 AA' that was logged in the serial monitor i should be able to put it into the mcucommand as a payload and see the change on the device. From there if it works a could easily work out the different modes for switching cooling, heating and ventilation. Is that correct?

Another thing your readme says "<YOUR_TOPIC>/mcucommand sends directly serial mcu commands to the device. The has to be a string in hexa-form without the checksum byte at the end" does this mean I have to ignore the last two characters from the serial commands I receive on the serial monitor?

ahmaddxb, 2019-05-01:
Tried to load my original firmware through the web server but it just stays on v0.9. Is that the right way to do it or should it be same as below.

4. Upload new firmware

Get the ESP in programming mode first. Erase flash:

esptool -p /dev/ttyUSB0 erase_flash

After erasing the flash, get the ESP in programming mode again. Write firmware (1MB)

esptool -p /dev/ttyUSB0 write_flash -fs 1MB 0x0 becaoriginalFirmware1M.bin

klausahrenberg, 2019-05-01:
Yes, the recorded commands should work as payload with mcucommand at my firmware. And yes, ignore the last 2 letters which you receive at the serial monitor.

klausahrenberg, 2019-05-01:
Ok, I thought, you can upload every binary, also the original one via web interface. If this not working, use esptool again as you mentioned. That was the way I did.

ahmaddxb, 2019-05-02:
By the way what are using to control the thermostat is it node-red or something like openhab or home assistant etc... Could you show me a screenshot of the GUI for controlling the thermostat.

ahmaddxb, 2019-05-02:
Had some success with the serial monitor. I found the commands to switch the AC modes (cooling, heating, ventilation) and have tested them in commandmcu with success.

I have made a dump of the serial command from the schedules in the attachment. The times of the weekdays and weekends are unchanged from the screenshot but i have set it at 25 degree and 26 degrees to see the different.
 
[commands.zip](https://github.com/klausahrenberg/ThermostatBecaWifi/files/3136877/commands.zip)

klausahrenberg, 2019-05-02:
Great, this helps a lot. After first view, it was impossible to determine these values in a 'logical' way. I will implement this tommorow

ahmaddxb, 2019-05-02:
Thanks again for you help.

I have a question about NTP time update i know when you first apply ac power to the device within 30 seconds it will update the time. How often does it do it afterwards?

You see the issue i have in my country is that weekdays are Sunday to Thursday and weekends are Friday to Saturday not the same as in the app. What i have done with the original firmware is just advance 1 day forward manually from the device and that was good enough to set my scheduled to reflect my country. 

If NTP is going to be updated frequently this will break my weekdays and weekends and put them out of sync for my country.

klausahrenberg, 2019-05-02:
Hi, I reviewed your data. Now it's clear why, you didnt received any unknown commands. I need to know 2 things to implement this correctly:
1. Your model has no external temperature sensor, right? You don't have 2 screw connectors and a external sensor with cable? Parameter 'actualFloorTemperature' is useless for you? Please confirm.
2. How are the modes working at the device. If I'm at my device not in manual mode, I can't set the desired Temperature. If you set at your device one of the modes cooling, heating, ventilation - can you set the desired temperature? For example: the room has 28 degree, your desired temperature is 22 degree, but you set your device to 'heating'. How does it work? Does the device switch to manual mode (hand symbol) and heats full power or can't you change the desired temperature in mode 'heating'?  I hope, my problem is clear, I need a small description of the behaviour to not implement impossible combinations in my source code.
Thanks in advance.

ahmaddxb, 2019-05-02:
1. No external sensor.

2. If im at my device in program mode i cant set temp on the device doesn't matter which mode im in (cooling heating ventilation) However i can still switch between these modes.

In manual mode i can set a temp in all working modes (cooling heating ventilation) and that set temp is shared between all modes. e.g I set temp to 25 in cooling and then switch mode to heating the set temp will stay 25. i dont have individual set temps for each mode.

klausahrenberg, 2019-05-03:
Version 0.91 should now support your device completely. Please test, see changes and readme for more details.
> about NTP ... the device within 30 seconds it will update the time. How often does it do it afterwards?

NTP and time zone sync is done every 3 hours 

> ... in my country is that weekdays are Sunday to Thursday and weekends are Friday to Saturday ... What i have done with the original firmware is just advance 1 day forward manually from the device ...

I implemented your solution with parameter 'schedulesDayOffset' in the same way. Set this to 1 via MQTT once. Value will be stored in device. Have a look in the mqtt state message and check the parameter 'weekend'. I 'schedulesDayOffset'==1, parameter 'weekend' will be true on Friday and Saturday and device is in weekend schedule (no. 6 or 7 on the LCD)

ahmaddxb, 2019-05-03:
Thanks again for this. Here are my results from testing.

Working 

schedulesDayOffset
weekend
thermostatModel detects BAC-002-ALW
actualFloorTemperature hidden in my model

schedules 0 reads my device schedules however if i change the schedules on the device it doesn't trigger a schedule mqtt message reflecting the changes (not sure if its supposed to work that way or not)

Sending a payload to schedules like below does trigger an mqtt message reflecting the changes

{
  "workday":{
    "0":{"h":"6:00","t":22}
  }
} 

Needs adjusting 

The heating payload triggers ventilation on the device
The cooling payload triggers heating on the device
The ventilation payload trigger cooling on the device

klausahrenberg, 2019-05-05:
> The heating payload triggers ventilation on the device ...  if i do a state 0 mqtt come back with "systemMode":"none"

Version 0.92 should fix the issue, there was a mistake in my header file with constant values

>  if i change the schedules on the device it doesn't trigger a schedule mqtt message reflecting the changes

This behaviour should not be the case: Normally you will receive your change back as a mqtt message and - if there was really a change at any value - you should get 3 messages with the whole schedule from the device. Please check, if you tried only to send the same change several times. If you really changed a value and received no response than your device didn't answer to a change - normally at every command the device responds with the same message back and this will create a MQTT message. In this case give me a message and I have to create a workaround for this.

ahmaddxb, 2019-05-05:
The systemmode payloads are all working correctly now with mqtt responses.

When I set cooling mode at the device it doesnt trigger an mqtt message.

I doubled checked setting the schedule at the device it does trigger an mqtt message. I changed many times and temps but as soon as it times out of programing mode nothing get sent over mqtt.

klausahrenberg, 2019-05-06:
> When I set cooling mode at the device it doesnt trigger an mqtt message.

> ...setting the schedule at the device it doesn't trigger an mqtt message.

Next try. Both problems above should be fixed now.

[7 more replies]

closed 2019-05-07 by klausahrenberg