klausahrenberg/WThermostatBeca #3: NTP answer can't be parced

a-roz, 2019-04-08:
I've put the binary supplied and got this message on MQTT broker.
I think that problem is escape "\" symbol.

{  "message" : "Can't parse json of time zone request result: {\"week_number\":\"15\",\"utc_offset\":\"+02:00\",\"unixtime\":\"1554725639\",\"timezone\":\"Europe/Madrid\",\"dst_until\":\"2019-10-27T01:00:00+00:00\",\"dst_from\":\"2019-03-31T01:00:00+00:00\",\"dst\":true,\"day_of_year\":98,\"day_of_week\":1,\"datetime\":\"2019-04-08T14:13:59.977447+02:00\",\"abbreviation\":\"CEST\"}"
}

a-roz, 2019-04-13:
Solved. Because answer of "http://worldtimeapi.org/api/ip" now get longer than @klausahrenberg suppose, you can repare this changing KaClock.cpp line 90 
StaticJsonBuffer<500> JSONBuffer;

klausahrenberg, 2019-04-13:
Thanks for the hint. Good that you find the fix already. I will correct this too in a new version.

klausahrenberg, 2019-04-29:
StaticJsonBuffer increased to 512 byte from version 0.7 on

a-roz, 2019-04-29:
I've forgot tell that's another problem when you inquiring current time.
1. Got time from NTP - have correct answer.
2. Adjust time by "http://worldtimeapi.org/api/ip". 
In a second step there is a problem. The NTP time is adjusted by value "utc_offset" from answer.
And after result is adjusting again if DST is active. 
Problem is "utc_offset" already give you a proper offset with DST. 
this mean if "utc_offset":"+02:00" and "dst":true , the real offset id +02:00.

"http://worldtimeapi.org/api/ip" answer confused us.

klausahrenberg, 2019-05-03:
Hi, I fixed it hopefully in version 0.91. Thanks for hint, please test and confirm because I'm living in an area without DST. Reason was that I used the Google API before. And there you had to add the utc_offset and dst_offset to get the local time. With "http://worldtimeapi.org/api/ip" it's even mor comfortable for me now. Please test.

closed 2019-05-07 by klausahrenberg

a-roz, 2019-05-07:
I can not check it now, as the equipment already installed and not available for reprogramming, hopefully I will check it later

klausahrenberg, 2019-05-07:
Yes, I'm quite sure that this is working now - so I closed this issue. Anyhow, try the actual code, during the work to support another model, I fixed a lot of other issues in the source.