klausahrenberg/WThermostatBeca #5: flash ok but... esp8266 bricked!

itMax86, 2019-05-09:
I have a bac-002-wifi board, some of the photo
I follow the step and all was ok, flash complete without error!
Then I try to connect but wifi icon doesn't appear! 
I try to reflash original fw but... it's impossible to connect to the esp8266
every time for every command I have

`esptool.py v2.6
Serial port COM3
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header`

itMax86, 2019-05-09:
Ok I found the problem!
It was a power problem! To avoid keep the Programmer connected via usb and disconnect only the VCC of the esp8266 module, then put GPIO0 to gnd and reconnect VCC
Now you can flash!

closed 2019-05-09 by itMax86

realthk, 2019-12-11:
Many thanks!
This should be in the flashing doc, it was also driving me nuts, resoldered every connection twice, but no use... Finally saw your comment, so instead of un- and replugging the USB programmer I put a switch on the power line to the thermostat, and there you go...

Strange, I've flashed many devices, and there was no problem with the USB connecting method, but not on this (and like in your case, for the first time it worked... but not now, flashing the second time with 0.97)