klausahrenberg/WThermostatBeca #9: idx Number

TwoLeftFeet, 2019-07-22:
I am using the Library successfully on a  BTH-002-GBLW central heating thermostat / controller.
I have managed to decode and send MQTT messages.
The inclusion of a "idx" number that is used in Domoticz included in your setup screen would be very useful.
I could then control the unit from within Domoticz.
Thanks for your work...

klausahrenberg, 2019-09-01:
I will added the idx already in the next bigger update - which will support Mozilla Webthings additionally. But it still need some time - it's to buggy at the moment.

klausahrenberg, 2019-11-15:
Sorry for late realization. Now the heating season starts, an update will come within the next weeks.

TwoLeftFeet, 2019-11-15:
Thank you very much for letting me know.....

TwoLeftFeet, 2019-12-09:
Thank you for the update.
I wonder if you had a chance to add the IDX number for use in Domoticz

klausahrenberg, 2019-12-09:
Yes, I'm already at testing, but it needs some more days

klausahrenberg, 2019-12-10:
Hi, I uploaded version 0.96. Please notice that a lot of things changed, e.g., MQTT topic structure. Your idx number is now added, but you have to reconfigure your topics maybe. Descriptions are not finished yet, I started here: https://github.com/klausahrenberg/WThermostatBeca/blob/master/Configuration.md

TwoLeftFeet, 2019-12-10:
Thank you very much for all your hard work have a lovely Christmas.

klausahrenberg, 2019-12-11:
Implemented since 0.96

closed 2019-12-11 by klausahrenberg

RoboPolaris, 2020-11-19:
Hello Klaus,

that all amazing work what you did with Beca. I would like to insert a buffer battery like a supercapacitor onto the bac-002 board which is part of the Beca Thermostat. I figured out that in the board (of the display with the microcontroller) there are 3 pins open, and 2 of it leads to VSS and VDD of the Mikrocontroller SONIX SN8F57084SG. Maybe this is dedicated to be the option to do exactly this. Do you have any experiences to electrical backup the system that after a loss of energy it is not necessary all the time to set up the system again? Thank you very much!

pery777, 2020-12-05:
Hey, can you please write me how you set up the domoticz to talk to the thermostat?  Where and which idx should be entered?  I would be grateful for help.