klausahrenberg/WThermostatBeca #11: BAC-002-ELW

mreyad, 2019-08-07:
I have thermostat BAC-002-ELW
[https://www.aliexpress.com/item/32916323547.html?spm=a2g0s.9042311.0.0.49ca4c4dFPXbFi](url)
I flashed the thermostat with your firmware, the steps to put the TYWE3S on flashing mode are slightly different than normal ESP8266.
I had to ground the "reset" pin two or three time while trying to connect using "esptool".
Note: The original firmware successfully backed up to my local PC.

After flashing with your firmware, I followed the steps and put the thermostat on pairing mode, I found the new AP and connected to that network.
When I tried to browse the IP "192.168.4.1", the AP disappears and I was not able to reach the page.

Then I tried to put the thermostat o pairing mode again, but it didn't respond to that action, although all other functions are still working (e.g. decreasing the temperature, configuring the scheduler, etc.)

Again flashed the device to the same firmware and same behavior happened.

I went through the code trying to understand how the device triggering the pairing mode, unfortunately i was not able, I am not a programmer so it took from me one week to change small thing in the code (after adding many Serial.print() lines) and I am able to get the AP again.

this time i used my NodeMCU to debug and understand how it is working.

Also with this way the same behavior I am getting from NodeMCU, the AP is up for some time, then disappears.

looking in the "serial monitor" I realized that if the NodeMCU is not able to connect to MQTT server it restarts the WiFi.softAP (which is the web-server)

Please let me know if you have some time to support or guide me on this.

RomanNagornyi, 2019-08-07:
Hello.

Maybe  I can help you. I have tested firmware for BHT-002- GBLW and BHT-002- GCLW. 

[image]

To transfer the module into flash mode, I needed to unsolder the module. Just be careful near the ADC pin, under it is a hidden track under the module. I do not know much about microelectronics, but it is possible that the reason in unsuccessful flashing. Try also erase the old firmware 

```js
esptool -p COM4 erase_flash
```
Hope it will help.

mreyad, 2019-08-07:
Hello Roman,
Thanks for your reply and help, but I was able to flash the chip multiple times, backup the original firmware, erase it and upload the custom firmware which is in this github.
The issue is after putting the thermostat on pairing mode, the wifi which is generated by the thermostat disappears before being able to browse the ip address "192.168.4.1" from the web browser.
And I am not able to put the device again on pairing mode, I had to flash it again to do this step

klausahrenberg, 2019-08-23:
Hi, problem sounds strange. WebServer should not stop after it was started. Furthermore you should start the "pairing mode" at the device more than once. 
The blinking lcd display with the wifi symbol is an independent device function - it has nothing to do with the flashed ESP module. The sequence is as follows:
- Device sends a serial command to the ESP that "pairing is requested" - the ESP starts the AP and you can connect to it with password "12345678", than the config page should popup.

Another strange thing is that you wrote:
>  i used my NodeMCU to debug and understand how it is working. Also with this way the same behavior I am getting from NodeMCU, the AP is up for some time, then disappears.

I would expect that no AP will be shown with a NodeMCU, because the initial request for pairing from the device would be missing.

If you already tried to build from source: At ThermostatBecaWifi.ino change line 18 from
`network = new KaNetwork(APPLICATION, VERSION, DEBUG, false);`
to
`network = new KaNetwork(APPLICATION, VERSION, DEBUG, true);`
Changing the last parameter means that the AP will start without no request of pairing mode, if no network settings are stored before. Flash it (to NodeMCO also possible) with option "Erase all flash content" to be sure that old settings will be deleted. The AP should start now. Keep in mind, after saving the AP will not come back. So configure WLAN and MQTT at the first time correctly. After access via MQTT you can start the WebServer via MQTT command. 
Maybe your device is not compatible yet. But if you can connect via MQTT, I could guide you to find out, which commands are send from your device to ESP. But first step is to connect the ESP to your network and MQTT.

mreyad, 2019-08-25:
Hello klausahrenberg,
Thanks for your reply and for your efforts.

I followed your instructions and made the change on "line 18" from false to true, then uploaded the same to the thermostat (TYWE3S).
Note: Already erased the firmware using esptool command.

I was able to see the WiFi "Thermos_Beca-Wifi_390537" and connect to using the password in the code "12345678", then was trying to browse the ip "192.168.4.1", but at that time the wifi disconnected and disappeared.

After that I tried with NodeMCU, uploading the code using Arduino IDE (without erasing, because I don't know) then the same behavior.

But with NodeMCU I was able to capture something using Serial Monitor and the following were repeated every time my mobile connected to the WiFi generated by NodeMCU.

I am going to keep trying and if I experience anything new, I will post back

> Exception (3):
> epc1=0x402152c4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x402490c3 depc=0x00000000
> 
> >>>stack>>>
> 
> ctx: cont
> sp: 3ffffc80 end: 3fffffc0 offset: 01a0
> 3ffffe20:  402490c3 3ffe8c65 3ffeffec 4020eec0  
> 3ffffe30:  3ffe8c65 3ffe8c65 3fff0754 40202803  
> 3ffffe40:  3ffffe68 3ffe8c65 3ffeffec 00000001  
> 3ffffe50:  3ffffea0 3fff0904 00000000 00000000  
> 3ffffe60:  ff000000 ffff1ffc 3ffffea0 40207248  
> 3ffffe70:  00000001 00000001 3fff08dc 402121f7  
> 3ffffe80:  00000002 40206690 3fff08dc 401000a9  
> 3ffffe90:  3fff08dc 3fff0030 3ffffec0 4020727e  
> 3ffffea0:  00000000 00000000 fffeffec 4020f234  
> 3ffffeb0:  3fff08dc 3fff0030 3ffeffec 40207309  
> 3ffffec0:  0000002f 00000000 00000000 00000000  
> 3ffffed0:  ff2f5054 0204a8c0 40100244 00006e07  
> 3ffffee0:  3fff0014 3fff00a0 00000001 40212040  
> 3ffffef0:  00000001 00000000 00000000 0000000f  
> 3fffff00:  00000000 3fff1afc 3ffeffec 3ffef044  
> 3fffff10:  00000001 3fff0014 3ffeffec 40207508  
> 3fffff20:  40212b50 00000000 00001388 4021a70c  
> 3fffff30:  00000000 3fff1afc 3ffeff6c 40208f46  
> 3fffff40:  00000000 00006e06 3fff0754 402030af  
> 3fffff50:  40104d60 00862e9e 3ffef0a0 00000000  
> 3fffff60:  3ffee7d0 3ffef0a0 3ffe8550 3ffef0a0  
> 3fffff70:  00000000 3ffef044 40210200 3fffefa0  
> 3fffff80:  402107fa 00000032 00000032 3ffef044  
> 3fffff90:  3fffdad0 00000000 3ffef014 40201566  
> 3fffffa0:  3fffdad0 00000000 3ffef014 402102b0  
> 3fffffb0:  feefeffe feefeffe 3ffe8550 401005d1  
> <<<stack<<<
> V+k⸮⸮J⸮

mreyad, 2019-08-26:
I flashed the NodeMCU with the official firmware v0.9 to make sure that I removed completely all the code uploaded, but still the same behavior.
I used "EspExceptionDecoder" to convert the stack that I mentioned in my last post:

> 0x4020eeb8: String::changeBuffer(unsigned int) at C:\Users\mreya\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\WString.cpp line 155
> 0x40204b7f: KaNetwork::handleHttpRootRequest() at C:\Users\mreya\AppData\Local\Temp\arduino_build_753889\sketch\KaNetwork.cpp line 300
> 0x4020aa08: FunctionRequestHandler::handle(ESP8266WebServer&, HTTPMethod, String) at C:\Users\mreya\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WebServer\src\detail/RequestHandlersImpl.h line 39
> 0x402122a7: std::__shared_count(__gnu_cxx::_Lock_policy)0>::~__shared_count() at c:\users\mreya\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\ext/atomicity.h line 67
> 0x40209e50: FunctionRequestHandler::canHandle(HTTPMethod, String) at C:\Users\mreya\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WebServer\src\detail/RequestHandlersImpl.h line 24
> 0x4010010d: std::function ::operator()() const at c:\users\mreya\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2/functional line 2465
> 0x4020f22c: String::operator=(char const*) at C:\Users\mreya\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\WString.cpp line 282
> 0x4020aac9: ESP8266WebServer::_handleRequest() at C:\Users\mreya\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WebServer\src\ESP8266WebServer.cpp line 606
> 0x40100244: millis() at C:\Users\mreya\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_wiring.cpp line 186
> 0x4020f779: String::equals(char const*) const at C:\Users\mreya\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/WString.h line 272
> 0x4020acc8: ESP8266WebServer::handleClient() at C:\Users\mreya\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WebServer\src\ESP8266WebServer.cpp line 310
> 0x4021a704: operator new[](unsigned int, std::nothrow_t const&) at ../../../../../dl/gcc-xtensa/libstdc++-v3/libsupc++/new_opvnt.cc line 34
> 0x40205423: KaNetwork::loop(bool) at C:\Users\mreya\AppData\Local\Temp\arduino_build_753889\sketch\KaNetwork.cpp line 138
> 0x402107f2: init() at C:\Users\mreya\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_wiring.cpp line 213
> 0x4020749a: loop() at C:\Users\mreya\OneDrive\Home Automation\tuya\Thermostate Firmware\ThermostatBecaWifi-master\src\ThermostatBecaWifi/ThermostatBecaWifi.ino line 71
> 0x402102a8: __yield() at C:\Users\mreya\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 103
> 

>

mreyad, 2019-09-01:
During the test I am doing on NodeMCU, I found the issue causing the error above.
I was using the library in Arduino for ESP8266 v2.5.2, after searching the internet for many hours I got to know that this was the reason, I downgraded the Library to 2.5.0 and it seems working for the first test.

I need to continue testing then apply the same on the Thermostat.

klausahrenberg, 2019-09-01:
Ok, thanks for feedback. I also use Library 2.5.0 - newer versions are not usable with sloeber which I used.

closed 2019-09-01 by klausahrenberg