klausahrenberg/WThermostatBeca #14: Webserver access boots the board

jsponz, 2019-09-05:
I have compiled the code. Just changes in the endpoint names. 

I am using platformio. 

I update the board and it works. I can use the endpoints with new names. I can even activate the webserver, but when accessing, board resets... 

Any idea?

jsponz, 2019-09-06:
Using the original code and I get the same crash. I have tested it with Arduino IDE and Platformio, and I get the following error:

Exception 3: LoadStoreError: Processor internal physical address or data error during load or store
PC: 0x40215284: umm_disconnect_from_free_list(unsigned short) at /Users/jorgesanchez/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/umm_malloc/umm_malloc.cpp line 1172
EXCVADDR: 0x40249083

Decoding stack results
0x4020ee7c: ESP8266WebServer::_parseRequest(WiFiClient&) at /Users/jorgesanchez/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WebServer/src/Parsing.cpp line 164
0x40204b6b: KaClock::loop() at /var/folders/r9/zmv5yzls5c7cjlvrchry37sc0000gn/T/arduino_build_921129/sketch/KaClock.cpp line 84
0x4020a9f4: ESP8266WiFiSTAClass::begin(char const*, char const*, int, unsigned char const*, bool) at /Users/jorgesanchez/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp line 145
0x4021226b: StreamString::write(unsigned char const*, unsigned int) at /Users/jorgesanchez/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/WString.h line 273
0x40209e3c: ESP8266WiFiAPClass::softAP(char const*, char const*, int, int, int) at /Users/jorgesanchez/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src/ESP8266WiFiAP.cpp line 170
0x4010010d: std::function ::operator()(String) const at /Users/jorgesanchez/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/xtensa-lx106-elf/include/c++/4.8.2/functional line 2465
0x4020aa2a: ESP8266WiFiSTAClass::begin(char const*, char const*, int, unsigned char const*, bool) at /Users/jorgesanchez/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp line 156
0x4020f1f0: ESP8266WebServer::_parseRequest(WiFiClient&) at /Users/jorgesanchez/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WebServer/src/Parsing.cpp line 237
0x4020aab5: ESP8266WiFiSTAClass::begin(char const*, char const*, int, unsigned char const*, bool) at /Users/jorgesanchez/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp line 92
0x4020acb4: WiFiClient::peekBytes(unsigned char*, unsigned int) at /Users/jorgesanchez/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src/WiFiClient.cpp line 299
0x40212b58: String::move(String&) at /Users/jorgesanchez/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/WString.cpp line 240
0x402099a3: onMqttCallback(String, String) at /Users/jorgesanchez/git/thermo2/ThermostatBecaWifi/src/ThermostatBecaWifi/ThermostatBecaWifi.ino line 131
0x4020c6c4: ESP8266WebServer::responseCodeToString(int) at /Users/jorgesanchez/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WebServer/src/ESP8266WebServer.cpp line 629
0x4020540f: KaNetwork::loadSettings() at /var/folders/r9/zmv5yzls5c7cjlvrchry37sc0000gn/T/arduino_build_921129/sketch/KaNetwork.cpp line 79
0x402101bc: HTTPClient::sendHeader(char const*) at /Users/jorgesanchez/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/WString.h line 289
0x402107b6: HTTPClient::writeToStreamDataBlock(Stream*, int) at /Users/jorgesanchez/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 1376
0x40207486: KaNetwork::startWebServer() at /var/folders/r9/zmv5yzls5c7cjlvrchry37sc0000gn/T/arduino_build_921129/sketch/KaNetwork.cpp line 269

lost-post, 2019-09-16:
Hi!
I tried the same and have the same result. When accessing webserver, the board resets...
I am using Arduino IDE with ESP8266 2.5.2. 
I am going to try to use 2.3.0. Some people say that Tasmota and so on works better on 2.3.0

jsponz, 2019-09-16:
Sorry, but I forgot to comment. 

Using "platform = espressif8266@2.0.4", it works :-)

closed 2019-09-16 by jsponz