klausahrenberg/WThermostatBeca #15: Replace tywe3s module?

cclein, 2019-09-07:
Hello, I have a tuya tywe3s based bac-002-wifi that was borked from factory (unable to go into pairing mode) . I have tried soldering to it directy and still nothing. Any reason why I can't desolder it and replace it with an esp-12?

TradeFace, 2019-09-22:
I think you can. See the link below. I broke my tywe3s too, now testing with esp-12f. So far all seems okay. 

https://github.com/arendst/Sonoff-Tasmota/wiki/Tuya-Smart-Plug-Models-ESP8266MOD-by-AI-THINKER-and-ESP-12F-by-TUYA-LN-0026

cclein, 2019-09-22:
I think you can too. Just waiting for a programming jig to show up and I will report back my success.

ninjaballs, 2019-11-08:
Were you able to successfully use an esp-12f in place of tywe3s?

TradeFace, 2019-11-08:
yes. they are interchangable. (at least in this case)

klausahrenberg, 2019-11-15:
Tywe module has only 1MB flash, ESP-12F mostly 4MB flash. Could be make a difference, I'm not sure.

closed 2019-11-15 by klausahrenberg

ninjaballs, 2019-11-15:
Nice, thanks for the update.  Seems from a pinout standpoint, the esp-12s might be a better match, going to give it a go

jsponz, 2019-12-09:
@TradeFace Could you confirm that it works? 

I have replaced the original TUYA module with an ESP12F and there is no communication with the thermostat MCU. ESP boots, but there is no communication. I can't interact with the thermostat.

TradeFace, 2019-12-09:
@jsponz It has been working without problems for a few months here. (BHT-6000)

Maybe you fried the MCU too?

jsponz, 2019-12-10:
@TradeFace Thanks for your answer. Yes, it was fried. 
I tried with another and it works!

nightmare04, 2020-09-23:
> Tywe module has only 1MB flash, ESP-12F mostly 4MB flash. Could be make a difference, I'm not sure.

Just replaced original module to ESP-07 work's fine.