klausahrenberg/WThermostatBeca #16: NTP Time is wrong

cclein, 2019-09-08:
Hello. I successfully updated my BHT-002-GBLW. I used TuyaConvert by the way and it worked perfectly. The unit connects successfully and the time is updated from 00:00. The time is 8 hours ahead. The time zone is correct with America/Thunder_Bay

tibo888, 2019-10-16:
Hello, i have the same problem
the time right now : 2019-10-16 11h24
clockTime : 2019-10-16 19h24

Someting else strange, lastNtpSync time is 2019-10-17 05h03

http://worldtimeapi.org/api/ip ->
{"week_number":42,"utc_offset":"-04:00","utc_datetime":"2019-10-16T15:26:16.457923+00:00","unixtime":1571239576,"timezone":"America/Toronto","raw_offset":-18000,"dst_until":"2019-11-03T06:00:00+00:00","dst_offset":3600,"dst_from":"2019-03-10T07:00:00+00:00","dst":true,"day_of_year":289,"day_of_week":3,"datetime":"2019-10-16T11:26:16.457923-04:00","client_ip":"XXXXX","abbreviation":"EDT"}

Did you find a solution?

uls169, 2019-10-21:
The similar problem here, time via mqtt is right but the display shows uptime of thermostat instead of current time. I rather tried to re-flash ESP, nothing changed.

tibo888, 2019-11-15:
Yes, the informations are all correct from worldtimeapi

tibo888, 2019-11-15:
I have a suggeston for a next release. Add a new field for disable request to worltimeapi like "IsNtpSync" and clockTime is an entry for callback

exemple: <YOUR_TOPIC>/IsNtpSync; payload: false|true
exemple : <YOUR_TOPIC>/clockTime; payload: "2019-15-11 12:04:26"

What do you think about that ?

cclein, 2019-11-15:
datetime is correct @ worldtimeapi for me

klausahrenberg, 2019-12-11:
Could you please install version 0.97 and test again?
Your suggestion is noted. You can configure NTP server and TimeZone Server in this release already. If you don't want time zone sunc, remove the entry.
Accepting clock times from utside of the thermostat is noted for next releases.

klausahrenberg, 2019-12-12:
New version 0.98 with several bug fixes. Please take attention to following: 
- After update all settings must be reconfigured
- MQTT topics has changed completely between v0.95 and later, see Readme.md for new json structure

closed 2019-12-15 by klausahrenberg

kovacsm, 2020-01-13:
Hi,
I have installed version 1.00, and I still have the same problem,
set the TZ to Europe/Budapest, which fits what I get on http://worldtimeapi.org/api/ip, but the clock is 1 hour back (i think it's UTC).
Just check my linux logs, where i capture MQTT messages:
**[2020-01-13 08:29:10]** thermostat/things/clock/properties {"ntpServer":"10.0.0.2","timeZoneServer":"**European/Budapest**","epochTime":1578900549,"epochTimeFormatted":"**2020-01-13 07:29:09**","validTime":false,"timezone":"","raw_offset":0,"dst_offset":0}

My other tasmota devices sync well from this server.
I'm a bit confused now it this issue fixed and I have some other problem, or... ?