klausahrenberg/WThermostatBeca #17: Relay states

uls169, 2019-09-15:
Hi, do I understand well that I can't get status of relay with this firmware? I don't want to set it manually, but it is good to see when thermostat's relay is on or off back in home automation software (hass in my case).
Thanks.

TradeFace, 2019-09-22:
with automation rules you could check if actualTemperature < desiredTemperature -> relay is on. 

How did you do the configuration in home assistant? Can you share your config for the themostat?

uls169, 2019-09-23:
Good idea with automation rule, but I think this way we could use as fault verification only.
Of course it is only an opinion. I don't know if the thermostat MCU sends state of relay to ESP. If not, my idea is nonsense.
I have not configured it in my production HASS server yet, I have this device installed for year with original FW, now I flashed klaus one and I only tried to check status and set it via mqtt (on my test HASS server). I asked for relay state in the moment when I thought about reasonable integration. Every relay device what I saw had state feedback (from sonoff to xiaomi or z-wave fibarro or heatit) if I remember well.
How I said, it's just idea, I don't know if it can be implemented well.

cclein, 2019-09-23:
I could see that being usefull. When the relay is on your home automation system triggers a ceiling fan, etc...

bikemike, 2019-11-09:
The MCU doesn't send the relay state. You would need to modify the board to hook the relay control pin on the MCU up to a gpio pin on the ESP. Then you'd need to modify the code as well.

I have done this and it works well but I'm using my own custom firmware.

Here is a pic of the board mod. I added a 10k/10k voltage divider to drop the MCU voltage from 5 to 2.5v.

[image]

pmaster555, 2019-11-09:
Can you post your firmware?

uls169, 2019-11-10:
@bikemike mike, thanks, simple idea, simple solution. Can you provide code diff so we can check your changes? Or firmware file as @pmaster555 mentioned in post above?

fabcia, 2019-11-13:
@bikemike,  I join the @pmaster555  and @uls169 requests for changes (diff in code) or firmware. I hope to have your news soon ....

bikemike, 2019-11-13:
I have some code that does this in my C17GH3 thermostat repository here:

https://github.com/bikemike/C17GH3Thermostat/blob/master/src/main.cpp#L35

Search for PIN_RELAY_MONITOR and relayOn

I'm adding support for my BHT-002-GBLW to that codebase but I haven't quite finished it yet. I'll post again when I get it working reasonably well with Home Assistant and the MQTT HVAC component.

klausahrenberg, 2019-11-15:
The device has no flag for relay working or not. I like the solution of bikemike. Maybe I will include the flag in my firmware also. At the tyua module are almost all IOs available, to use this for relay state is a good idea.

bikemike, 2019-11-15:
I've added support for BHT-002-GBLW to my firmware now but be warned that it isn't very user friendly. It requires changing the timezone in code and compiling - both of which I haven't documented yet.

https://github.com/bikemike/MikesThermostat

coldfire84, 2019-11-22:
Is there something we can do here that's purely software? i.e. some logic in the formware that works out target temperature, set point, whether heating schedule is active or manual mode is enabled and returns what is, **in effect**, the state of the relay?

@bikemike, does your hardware modification allow for **control** of the relay, or is it simply state only?

pmaster555, 2019-11-22:
> is it simply state only

state only

bikemike, 2019-11-22:
You could probably compute the state of the relay but I don't think the "deadzone temperature" setting is transmitted from the MCU so you would have to have an additional setting for that on the wifi module. For me it was just easier to solder a wire.

If you want to control the relay then you would have to make additional hardware changes. I did this for another thermostat I have (C17GH3) so I could control two relays (for baseboard electric and a gas fireplace). I disconnected the MCU from the relay and rewired so the wifi module reads the state from the MCU and controls both relays.

valepe, 2019-11-29:
Does anyone can tell me where solder the voltage divider on the BHT-6000 thermostats?
Link: [BHT-6000 board](https://github.com/arendst/Tasmota/issues/5077)

bikemike, 2019-11-29:
@valepe You will need to check which pin on J1 connects to the relay control. If it is like my board(bac-002-wifi), the second from the bottom is GND, and 4th from the bottom is the relay pin.

klausahrenberg, 2019-12-11:
Software solution for this hw modification is implemented now in 0.97

closed 2019-12-11 by klausahrenberg

ahmaddxb, 2019-12-11:
> 
> 
> Software solution for this hw modification is implemented now in 0.97

Can you elaborate more on this? Do you still need to do the hardware mod or is this all done in software now.

klausahrenberg, 2019-12-11:
Hi, ok to explain. This issue can not be solved by software only. The thermostat (MCU) definitily sends no state info abut the heating relay. Please do the small modification that bikemike showed here: https://github.com/klausahrenberg/WThermostatBeca/issues/17#issuecomment-552078026
My firmware now asks for GPIO05 and sends the state. But it works only with this hardware modification.

ahmaddxb, 2019-12-12:
@bikemike My board look like this. As you can see I have a J3Y on Q3 and Q4 as well as something on R9 and R10. How would I go about checking which connection point is needed to be made. Could i test is with a multimeter while turning on and off the relay by adjusting the temperature controls on the unit? Also what would I be looking for? Or could you tell me the correct way to approach this. Thanks

[image]

realthk, 2019-12-12:
It is a different setup indeed, though the board is the same, but your seems to have transistors where our version is left empty at Q3, Q4, only a 0 resistor on Q4:
[image]
But as the panel is the same, the relay control line is most likely still on R10 and the affected leg of Q4 is also GND, so it should not be any different to get the signal.

klausahrenberg, 2019-12-13:
> My board look like this. As you can see I have a J3Y on Q3 and Q4 as well as something on R9 and R10.

**@ahmaddxb** Could you let me know your exactly model? Does your model supports more function, like cooling/ventilation? And also, if the hw modification is also working at your model - I would wonder, if this would not work because layout is the same.

ahmaddxb, 2019-12-13:
Yes mine has heating cooling and ventilation modes, model is BAC-002ALW used for 2 Pipe 3 Speed Central Air conditioner. 
I only use it for cooling side and wanted to monitor the state of the repay to know then the valve is opened on my AC.

https://www.aliexpress.com/item/32913037607.html?spm=a2g0s.9042311.0.0.27424c4dqwOdze

I haven't tried the hw mod yet i wanted to know a way to test if putting the wire in the same location as @bikemike if it could work on my model. Is there some test i could do with a multimeter while triggering the relay on and off with the unit by adjusting the temperature to above and below the desired temp.

bikemike, 2019-12-14:
@ahmaddxb, yours probably has multiple relays so you will have to figure out which one is used for cooling and trace that back from the relay to the board. You'll have to disconnect the thermostat and use a multimeter to check continuity with the relay signal pin and the ribbon cable pins.

shin-01, 2019-12-21:
For some reasons, I think in BHT-6000 heating and off state are swapped...

OFF state:
{"idx":"thermostatbeca_11482470","ip":"192.168.1.105","firmware":"1.00","temperature":20.50,"targetTemperature":20.00,"deviceOn":true,"schedulesMode":"off","state":"heating"}

On State:
{"idx":"thermostatbeca_11482470","ip":"192.168.1.105","firmware":"1.00","temperature":20.50,"targetTemperature":22.50,"deviceOn":true,"schedulesMode":"off","state":"off"}

funtik039, 2020-05-10:
[image]
[image]

Hi all. I have BAC-2000WIfi. Its the same as BAC-002GALW. But it have another board.
Some one can help whis it? How to upgrade this termostat?

jodelflyer, 2020-06-04:
I have been using your firmware for about 6months and it is working well.  However, I want to cjhange the IP address it connects to.  I have tried accessing it via the http://<device_ip>/config (using my IP) but it does not open.  The state from the thermostat says "webServerRunning":false.  I assume tis is the problem.  How do I make it true

klausahrenberg, 2020-06-04:
Ok, it seems you use a really old version, if the webserver is not running all the time. It changed from v1.00 on. If you want to update, keep in mind that mqtt topics also changed - take a look to the readme in this case.
To activate the web Server. Switch thermostat off and press the 'down'-Button until the display blinks. Then the thermostat is in config mode and should start the webserver. Maybe it opens an own Wifi-Acces-Point, but normally it just starts the webserver and is available under http://<device_ip> at this old version.
Please check. Alternativally send a mqtt message <your_topic>/webServer and payload 'true' - this should start the config page too.

jodelflyer, 2020-06-04:
Perfect.  Thanks for the very quick response.

Tony

jodelflyer, 2020-06-12:
Following on from the above, I had to reset my node red control of my thermostat after upgrading the firmware (even though you warned me about topic changes I did not see the full changes).  Generally everything is working fine.  However, I do not have any inputs to change the scheduleMode but find that the thermostat keeps changing to auto mode.  All of my thermostat controls are in one flow.  I have disabled it but the thermostat is still switching to auto.  Is there an issue.

fyodorov-alexey, 2020-06-12:
Hello everybody!
If I understood right, this thermostat sets relay to start floor heating when air temperature on its body sensor is lower, than set as desired.
Is there a possibility to bind the floor heating to the *floor sensor* temperature?
That is very important for me in my use case.
Alexey

ReachyBayern, 2020-06-14:
Hardware change to get "heating" state for Beca Wifi thermostat (Kkmoon E8489W-2)
[image]

R13 - R10K - GPIO5 - R10K - GND

GND is also there where the black mark is. But the chassis allows there less space. 

Kind Regards

Roland

ReachyBayern, 2020-06-14:
> 
> 
> Hello everybody!
> If I understood right, this thermostat sets relay to start floor heating when air temperature on its body sensor is lower, than set as desired.
> Is there a possibility to bind the floor heating to the _floor sensor_ temperature?
> That is very important for me in my use case.
> Alexey

yes, you can use the external sensor. see manual

[https://irshop.eu/eng/files/2017/11/BHT1000-WiFi-Eng-irshop.eu_.pdf]

[image]

fyodorov-alexey, 2020-06-17:
> yes, you can use the external sensor. see manual
> 
> [https://irshop.eu/eng/files/2017/11/BHT1000-WiFi-Eng-irshop.eu_.pdf]
> 
> [image]

Thank you very much for manual link! My thermostat slightly differs, and service menu differs too. The external NTC sensor selection is "OU".
[image]
BTW, when NTC is chosen as a target sensor, MQTT **temperature**, posted by thermostat equals to **floorTemperature**. it’s bit a pity that can’t use the thermostat as an additional air temperature sensor.

GGW42, 2020-08-20:
Hi, I have just flashed my "BHT-002" "GCLW" with V1.11. 
Is the web interface really a text page? See attached image.
[image]
There doesn't seem to be a way to change any settings from this page, is this correct or have I made a mistake somewhere?
I saw the graphic web interface just after flashing when it was an access point but have not been able to find it again. I want to add this mod to monitor the relay but am not sure how I can turn it on since there are no controls on my web interface?
Any clues, i am really new to this so i probably missed something somewhere?

fashberg, 2020-08-20:
@GGW42 
Try: http://192.168.0.203/config 

https://github.com/klausahrenberg/WThermostatBeca#initial-configuration

Initial configuration
To setup the device model, network options and other parameters, follow instrcution here:
https://github.com/klausahrenberg/WThermostatBeca/blob/master/Configuration.md
After initial setup, the device configuration is still available via http://<device_ip>/config

GGW42, 2020-08-20:
Thanks @fashberg not sure how i missed that, it works well.
I have now moded my unit so the "state" function now works. My unit has a PCB marked "bac-002-wifi-20190409" but looks the same as the "bac-002-wifi" unit above so I followed that mod. I have now installed unit on the wall and it continues to work well. Now I just have to work out how to get "Home Assistant" to see the MQTT messages that I can see being transmitted.

fashberg, 2020-08-20:
Hey @GGW42 
For homeassistant i suggest my fork and then just use autodiscovery.
Use latest https://github.com/fashberg/WThermostatBeca/releases/tag/v1.16.beta3-fas which has also autodiscovery for BAC-002 climate control.

You can see also HASS manual config at my project startpage.

I currently have 2 problems with BAC-002
HASS supports only these modes:  [“auto”, “off”, “cool”, “heat”, “dry”, “fan_only”]
But BAC-002 has no full-auto-mode, you can switch between auto (scheduler) and manual and the mode can be switched between cool/heat/fan_only. How to map this to HASS. Any suggestions?
Bug2: when changing fan-speed mode switches to fan_only

Please reply to https://github.com/fashberg/WThermostatBeca/issues/38 for fork-specific 

Kind Regards
Folke

GoNzCiD, 2020-09-28:
> You could probably compute the state of the relay but I don't think the "deadzone temperature" setting is transmitted from the MCU so you would have to have an additional setting for that on the [...]

Hi @bikemike, can you share what hardware mod is needed to control the relay with mqtt? Thanks

rumoldus, 2020-10-14:
> The MCU doesn't send the relay state. You would need to modify the board to hook the relay control pin on the MCU up to a gpio pin on the ESP. Then you'd need to modify the code as well.
>
> I have [...]

Hi bikemike, could you describe in slightly more detail where you connect your resistors ? I think I see the open pad of Q4 and the nearby end of R10, but it is not entirely clear to me. Many thanks.
Marcel

fashberg, 2020-10-14:
@GoNzCiD 

I have implemented RelayStateCalculation.
The DeadZone configured in the Thermostate-setup (MCU) must be configured in WThermostat then you get the state

https://github.com/fashberg/WThermostatBeca

bikemike, 2020-10-16:
> Hi @bikemike, could you describe in slightly more detail where you connect your resistors ? I think I see the open pad of Q4 and the nearby end of R10, but it is not entirely clear to me. Many thanks.
> Marcel

Yes, one resistor connects to the open pad and the other resistor connects to the end of the 0 ohm resistor on what in guess could be R10(I dont see a label for it though).

The other ends of the two 10k resistors are soldered together and that is where you solder the wire to as well.

GoNzCiD, 2020-10-18:
@fashberg, thanks. I was trying to control the relay, not only check the real state. Do you know if it's possible and the needed hardware mod?

Thanks

bttfw, 2020-10-24:
> Hi all. I have BAC-2000WIfi. Its the same as BAC-002GALW.
> [image]
> 
> But it have another board.
> Some one can help whis it? How to upgrade this termostat?
> [image]
> 
> In my logic it must be like this:
> [image]

Did you ever figure it out ??

gavlexx, 2020-11-28:
Mine BHT002-GBLW board is like this.
Found that when relay is on - there is 5V between 1 and 2 (from top) pins of J2. Otherwise 0V. 
Getting 5V through divider to GPIO pin and have profit! :)
[image]

Freddyjb67, 2020-12-03:
Welcome!

Before I burn,
BHT-6000 ..
it is possible to operate in Domoticz.

Thanks!

frtz13, 2020-12-20:
> Mine BHT002-GBLW board is like this.
> Found that when relay is on - there is 5V between 1 and 2 (from top) pins of J2. Otherwise 0V.
> Getting 5V through divider to GPIO pin and have profit! :)

I got the same board (model : BHT-002-GCLW), and had a closer look at J2. pin 3 is GND. Pin 2 is 0V when heating, 5V when heating is off.
I put a voltage divider (2 x 15kOhm) between pin 2 and 3 and connected the middle point to GPIO5.
[image]

However, in Wthermostat 1.19beta1-fas with configured heating relay connection at GPIO 5, I get the inverted heating state on the Info page, and through MQTT.
The inverted state through MQTT is not a big problem if you use it in Home Assistant. The state can be inverted again in the climate configuration section:
```
climate:
  - platform: mqtt
(…)
    action_template:  >
      {% if value_json.action == "heating" %}
        idle
      {% elif value_json.action == "idle" %}
        heating
      {% endif %}
(…)
```
@fashberg  Maybe in a future release we can get one more configuration option in the drop-down list: heating relay connection at GPIO 5 (low=active)…

gavlexx, 2021-01-01:
```
I got the same board (model : BHT-002-GCLW), and had a closer look at J2. pin 3 is GND. Pin 2 is 0V when heating, 5V when heating is off.
I put a voltage divider (2 x 15kOhm) between pin 2 and 3 and connected the middle point to GPIO5.
```

If you put voltage divider between pin 1 and 3 of J2 - you receive non-inverted heating state :)

frtz13, 2021-01-02:
> voltage divider between pin 1 and 3 of J2
pin 3 is GND, at same potential as the ESP8266 GND pin. according to my measure, pin 1 is 5V, regardless of relay state (can't do it again, thermostat is now taking care of the heater...). voltage divider between pins 1 and 3 does not make sense to me, at least not with the signals on my board. btw, thermostat is working fine and the (inverted) relay state signal looks good with my pin 2/3 divider.

gabor60i, 2021-01-17:
> > Mine BHT002-GBLW board is like this.

@frtz13: My board exactly the same. Sadly I have no experience in soldering so originally I planned to use tuya convert. As turned out it is not possible with this new tuya fw. So I would like to give it a try with soldering, but not find any reference picture of how to do it.
I would appreciate if You share with me a picture of how to connect the pins.
Thanks!

frtz13, 2021-01-17:
wiring info is here: https://github.com/klausahrenberg/WThermostatBeca/blob/master/Flashing.md
good luck.

gabor60i, 2021-01-17:
Do You mean, that regardless of the differences in the board, and the name of the components, the wiring is the same?
I found this guide myself, but I was disappointed because the differences was too obvious.

frtz13, 2021-01-17:
yes, wiring is the same. it's the same ESP chip. I used this guide and everything worked as expected.

gabor60i, 2021-01-18:
Thanks for Your quick answers!
I will give it a try.

GUVWAF, 2021-02-08:
Just for reference: I got the trick of @bikemike working for the ME81H thermostat. For this device, the voltage at R2 is 5V when the relay is on and 0V when it is off. So, using a voltage divider you can read the state using GPIO5 of the ESP chip.
[image]

znanev, 2021-02-14:
> Did you ever figure it out ??

@bttfw I have this same model. Thinking about modifying it to be able to determine the relay status.

Following are my observations:

When the relay is switched off, the terminals of the 0-ohm resistor R14 measure 5V (compared to ESP ground). When the relay is swithced on - the terminals of R14 measure 0V (same potential as ESP ground).

So in order to obtain non-inverted input of the relay state, the voltage divider should be pulled up to +5V (instead of pulled down to GND as shown in @funtik039's picture).

I traced some of the signals to the 3-pin unpopulated connector on the left - the central one is GND - the same potential as ESP's ground, the one below it is +5V and it can be used for the voltage divider:

[image]

Please note that I still haven't modified my board, just wanted to share my observations. I'll probably test this modification next week.

MadDoct, 2021-02-16:
> @bttfw I have this same model. Thinking about modifying it to be able to determine the relay status.
>
> Following are my observations:
>
> When the relay is switched off, the terminals of the 0-ohm [...]

Aren't you injecting 5v directly to the esp pin when R14 is at 5v?! That's not a voltage divider...

martin-vi, 2021-02-16:
My BHT-002 GALW has yet another board layout, labeled "bac-002-wifi-20190409". New firmware, Tuya Convert did not work. WThermostatBeca is now running with firmware 1.19, Version 1.20l only worked with strange "Unknown MCU command" errors, maybe https://github.com/klausahrenberg/WThermostatBeca/issues/186.

 Reading the relay state with ESP12 GPIO 5 from R10 with 10k/10k voltage divider works well (https://github.com/klausahrenberg/WThermostatBeca/issues/17#issuecomment-552078026).
Some additional notes: I only had to connect VCC, GNG, RX, TX and GPIO0 to enter programming mode.

[image]

znanev, 2021-02-16:
> Aren't you injecting 5v directly to the esp pin when R14 is at 5v?! That's not a voltage divider...

Oops, that's exactly what would happen... 5V on the ESP and magic smoke, maybe :) Thanks for pointing this out, @MadDoct!

Since I've posted this, I took some other measurements and found the non-inverted signal on R13.

realthk, 2021-02-16:
> Oops, that's exactly what would happen... 5V on the ESP and magic smoke, maybe :) 

No, the ESP is 5V tolerant on its input pins.

martin-vi, 2021-02-16:
> No, the ESP is 5V tolerant on its input pins.

@realthk are you sure, any sources on that? I only found https://hackaday.com/2016/07/28/ask-hackaday-is-the-esp8266-5v-tolerant/. I would stay at 3.3 V to be on the safe side.

[44 more replies]