klausahrenberg/WThermostatBeca #18: Flash question

TradeFace, 2019-09-19:
Hi Klaus,

Thank you for this repo. I have a BHT-6000-GBLW which I am trying to flash with your bin file. I followed your instructions by the letter, but sadly I keep getting 

> A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header

After running:

`sudo esptool -p /dev/ttyUSB0 -b 115200 read_flash 0x00000 0x100000 originalFirmware1M.bin`

I tried other wiring schemes: without io15, without EN; other baudrates; with RST on boot, with RST and IO0 on boot. Nothing seems to work. 

I am missing some magic on booting the device. Do you have some pointers? 

Thanks! 

Best regards,

Niels

egony, 2019-09-20:
Hi!

I also could not flash the device (BHT-002-GBLW) for a long time. I think the MCU of the thermostat (rx-tx) inhibit flashing.

I had to unsolder the wifi module (TYWE3S) to flash. In this case, it is enough to connect rx, tx, 3.3v, GND to the programmer, and connect the switch for GPIO0-GND. Another condition for flashing of the wifi module is the presence of a connection between EN and 3.3v.

Maybe you case is same...

TradeFace, 2019-09-20:
@egony Thanks for the info! 

I did desolder the module (after posting here). esptool now does detect the module and puts it in flashmode but doesn't want to read it. 

`$ esptool  -p /dev/ttyUSB0 -b 115200 --before default_reset --after hard_reset read_flash 0x00000 0x100000 originalFirmware1M.bin
esptool.py v2.1
Connecting...
Detecting chip type... ESP8266
Chip is ESP8266
Enabling default SPI flash mode...

A fatal error occurred: ESP8266 ROM does not support function read_flash.`

Should you keep GPIO0-GND connected during the whole flashing process?

egony, 2019-09-20:
Procedure for entering flash mode:
Short GPIO0-GND, reset the module (in this moment I just connect my programmer to USB and wait for 2-3 sec), open GPIO0-GND circuit.

I use these parameters:
esp_tool -pCOM3 -b115200 -fmDOUT -a0x0000 -s0x100000 -or esp82XX.bin
(esp_tool.exe, windows OS)

RomanNagornyi, 2019-09-20:
I also desoldered the wifi module. https://github.com/klausahrenberg/ThermostatBecaWifi/issues/11#issuecomment-519150072

TradeFace, 2019-09-20:
The read_flash problem was probably caused by an older version of esptool (2.1); next to the that is was soldered on to the pcb.

With version 2.7 I finally was able to read.

```
esptool.py  -p /dev/ttyUSB0 -b 115200 read_flash 0x0000 0x100000 tuya-thermostat-bht-6000-gblw-2.bin
esptool.py v2.7
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: bc:dd:c2:40:43:7d
Uploading stub...
Running stub...
Stub running...
1048576 (100 %)
1048576 (100 %)
Read 1048576 bytes at 0x0 in 94.6 seconds (88.7 kbit/s)...
Hard resetting via RTS pin...
```
I guess the hard part is over (wishful thinking :-) )

TradeFace, 2019-09-20:
Well, I was able to flash it. But the wifi never came up. 

Now I flashed Tasmota onto it, which does kinda work. With the tuya dimmer template selected I am able to set temperature, switch on/off the device, eco mode and locking. Maybe with some additional config I might be able to read the temperature (ambient, floor). 

There I go down into the rabbithole.

Thanks for the tips!

closed 2019-09-20 by TradeFace

MiklerGM, 2021-10-10:
I had the same problem with esptool installed from repository on Ubuntu.

The problem was solved once I've tried esptool.py install via pip

bruno-dogancic, 2022-10-01:
> I had the same problem with esptool installed from repository on Ubuntu.
> 
> The problem was solved once I've tried esptool.py install via pip

Thank you so very much. Saved me a day! Was almost ready to de-solder the board :D