klausahrenberg/WThermostatBeca #23: Relay On/Off

seblo0505, 2019-10-09:
Hi, i think your work is very great. I wish I could see the State of the relay but this is not possible. So i want to controll the state with the display. I can Power it off with "YoutTopic/deviceOn 0". But i cant it Power on with "beca/deviceOn 1" or "beca/deviceOn On"; On the Broker I see that the message has arrives with 1. Can you help me?

Best regards

coldfire84, 2019-10-09:
> I can Power it off with "YoutTopic/deviceOn 0"

I don't think this is powering the relay itself, rather the screen/ display? [deviceOn is a boolean](https://github.com/klausahrenberg/ThermostatBecaWifi#state-record---json-structure), so use true/ false not 0 or 1.

I agree that being able to view and control the relay state would be really useful. I have multi-sensors across the house that could be used to trigger heating in a "cold" room where the Thermostat itself is not present. Of course, I could simply increase the setpoint, but this feels like a hack.... :)

seblo0505, 2019-10-10:
No it doenst power the relay, but the screen, and when the screen is off, the relay doesnt trigger. I integrated it in my smarthome, and every time i dont want to heat I turn the display off.

TradeFace, 2019-10-11:
> I have multi-sensors across the house that could be used to trigger heating in a "cold" room where the Thermostat itself is not present.

You can add a sonoff basic, of which you can control the relay, for multiway switching.

coldfire84, 2019-10-11:
I'd considered using a Sonoff Mini (can hide behind the stat). Just wanted to avoid 2x WiFi devices controlling my heating in one zone.

closed 2019-11-15 by klausahrenberg