klausahrenberg/WThermostatBeca #24: BTH-6000-X or BHT-002-X (but NO wifi enabled versions) can be flashed flashed with this firmware ???

jirmjos, 2019-10-11:
Good day all ! .

I want to know if anyone had try to reflash with this firmware any of the beka thermosthat models BHT-006-x or BHT-002-x but NO wifi/tuya enabled versions ??

Had anyone a non wifi enabled version and can open it to see if TYWE3S communication module is still there in that non wifi versions ???

And last, anyone knows if that NO wifi versions is the modbus still poppulated and working ???

Best regards,

klausahrenberg, 2019-11-15:
Hi. Mi thiunking about this: 
1. No WIFI enabled version: I'm quite sure that the tywe3S-module is not mounted in this versions, maybe it's also a complete other PCB layout. If ther would be an empty space for the TYWE3S module, you could mount an ESP-12F module on it and flash this firmware. However, I expect that the device firmware will not communicate with the ESP module - that would be to easy. 
2. Modbus device: I know that from own experience. If you didn't take attention when you bought the device, you will not have a modbus - it's an additional feature. I tried to solder the modbus-connection afterwards - but there was no communication possible with the device.

closed 2019-11-15 by klausahrenberg

velijv, 2023-01-04:
Maybe someone will stumble on this question and be happily surprised: https://community.home-assistant.io/t/convert-dumb-thermostat-into-a-smart-one-hy08/497105