klausahrenberg/WThermostatBeca #25: How to get the status On?

alleonname, 2019-10-13:
Hi! Great Work! Thanks!
How to get the status On to HA?
My config to configuration.yaml:
`climate:
  - platform: mqtt
    name: Study
    qos: 0
    payload_on: On
    payload_off: Off
    modes:
      - 'off'
    power_command_topic: "study/ac/power/set"
    mode_command_topic: "study/ac/deviceOn"
    current_temperature_topic: "study/ac/state"
    current_temperature_template: "{{ value_json.actualTemperature }}"
    temperature_command_topic: "study/ac/desiredTemperature"
    initial: 29`

[image]
[image]

If you press the on/off button while the thermostat is on, it will turn off. What needs to be fixed so that the status changes to ON?
If you press the Power button on the Thermostat itself, the status changes: false/true.
[image]

TradeFace, 2019-10-16:
I don't think you can switch it on with the current firmware. If I remember correctly I could switch it (BHT-6000-X) on with tasmota. (But a whole lot of other functions didn't work). 

Did you try to do mqtt publish for the deviceOn topic? 

Anyways, here is my config so far, maybe it helps in some way: [https://github.com/TradeFace/haconfig/blob/master/packages/living_thermostat.yaml](url)

alleonname, 2019-10-16:
broken link "No results matched your search"

iMike78, 2019-10-16:
> Anyways, here is my config so far, maybe it helps in some way: [https://github.com/TradeFace/haconfig/blob/master/packages/living_thermostat.yaml](url)

Wow, thank you! It's working for me ;)

TradeFace, 2019-10-17:
> I don't think you can switch it on with the current firmware. 

I was mistaken. You can publish to `<mqtt topic>/deviceOn` true|false And it reacts as expected. But indeed the on/off button on the climate.mqtt card isn't working correctly.  Maybe it can be captured as an event or with an automation?

alleonname, 2019-10-17:
I'll check tonight, but it seems like this should work correctly:
payload_on: "true"
payload_off: "off"

TradeFace, 2019-10-17:
Yep, you're right. I've added 
```
 payload_on: "true"
 payload_off: "false"
``` 
now it can be switched. Though switch the auto mode in HA is still a problem.

alleonname, 2019-10-18:
> Anyways, here is my config so far, maybe it helps in some way: https://github.com/TradeFace/haconfig/blob/master/packages/living_thermostat.yaml

alleonname, 2019-10-18:
@TradeFace,
thank you very much, it works
I installed ESP-12 instead of TYWE3S. For some reason, the temperature readings are 10 gr more. But this is most likely due to my installation error.
I ordered the programmer on AliExpress. Waiting for delivery. I want to try reflashing TYWE3S without removing it from the board.
[image]
[image]

TradeFace, 2019-10-18:
>  the temperature readings are 10 gr more. But this is most likely due to my installation error.

Hmmm I have the same problem with my thermostat. I thought I partly broke it with pulling the connector out while the thing was still connected to the grid. Well at least I broke something; the external sensor for floorheating is always at 107 degrees or something like that. I got a replacement in yesterday. So, I can have another go at it ;-) In the menu of the device you can do a correction of the actual temperature from -7 to 9 (on BHT-6000)

I couldn't get it flashed with the wifi module on the board. Maybe also due to the lack of experience.

alleonname, 2019-10-18:
I also deleted a little vandal )))
[image]

mrgluek, 2019-10-18:
Your need to have 10k sensor, if you connected to 7k sensor your already have you need to add some resistors (like +1k+2k) to get 10k total.

alleonname, 2019-10-18:
Ok. Thanks!

nimdy2017, 2019-11-11:
> Anyways, here is my config so far, maybe it helps in some way: https://github.com/TradeFace/haconfig/blob/master/packages/living_thermostat.yaml

Thank you for this. I've copied this into my own configuration. I've also changed the mode_state_template to show whether the thermostat has been put into scheduled mode on the thermostat itself. 

I've noticed my Nest thermostat has an idle status. The heating setting is on (little orange flame symbol) but the status in the Lovelace thermostat shows 'Idle'. Does anyone know how we can implement this in the Generic Thermostat configuration?

closed 2019-12-11 by klausahrenberg