klausahrenberg/WThermostatBeca #27: WiFi Manager

31warren, 2019-10-17:
Thanks for sharing this code - works really well.

I have flashed a BHT-002-GCLW branded Moes using the 0.95 binary.

The device connects to WiFi and sends/receives MQTT commands ok.

The WiFi symbol is not showing up on the display even though I have a connection (not a problem as I have MQTT connectivity). I am able to turn the web server on over MQTT but this does not seen to activate an access point. Turning the device off and entering discovery mode also doesn't seem to activate an AP.

Is there any way to turn on the WiFi Manager configuration page again - I would like to move the thermostat to another part of the house that has a different WiFi SSID.

Thanks.

iMike78, 2019-10-17:
> Is there any way to turn on the WiFi Manager configuration page again - I would like to move the thermostat to another part of the house that has a different WiFi SSID.
You can start the webserver. Publish an mqtt message (for example with MQTT Explorer) to this topic with "true" payload:
<YOUR_TOPIC>/webServer/
In my case:
home/nappali/thermostat/webServer
[image]
[image]

31warren, 2019-10-17:
Thanks for the reply.

I have tried this but no new access points are showing up.

Here's my config...

{"deviceOn":true,"desiredTemperature":18.5,"actualTemperature":20.5,"actualFloorTemperature":0,"manualMode":true,"ecoMode":false,"locked":false,"thermostatModel":"BHT-002-GBLW","logMcu":false,"schedulesDayOffset":0,"weekend":false,"clockTime":"2019-10-17 21:10:11","validTime":true,"timeZone":"Europe/London","lastNtpSync":"2019-10-17 20:23:14","firmware":"0.95","ip":"192.168.0.127","webServerRunning":true}

iMike78, 2019-10-17:
> 192.168.0.127

Type this to your browser address line, and the setup page will be there ;)

31warren, 2019-10-17:
Thank you, worked ok.

closed 2019-11-15 by klausahrenberg