klausahrenberg/WThermostatBeca #28: Odd BAC-002-WIFI variant

cclein, 2019-10-21:
[image]

Attached is an image of the thermostat I received from MoesHouse. I was never able to get this thermostat to get into wifi pairing. Moes was quick to send me a replacement which was physically different from this original one. I'm wondering if anyone has seen this unit before and has gotten it to work with this firmware. I could never flash the firmware so I removed the tuya module and flashed external. It never gets the correct time even though I can ping it etc... Every day or so I get an mqtt error message from it.

Just wondering if I can salvage this thing.

klausahrenberg, 2019-11-15:
I also have one device from MoesHouse which works correct. Your unit looks like mine, except the hand-soldered resistor/diode? That sometimes the time sync sometimes fails, is normally not a problem, if the unit synced once successfull before. Device will try after some hours again to sync. Does the device synced the time before - or was it never showing a correct time?

cclein, 2019-11-15:
No, never showed correct time. Took about 24 hours to get any mqtt messages, which was an error.

cclein, 2019-11-19:
And to add, I could never pair it with the app.

cclein, 2019-11-19:
It was a dud from the get go...I was going to use it to to test 240V AC here in Canada. We have split phase AC here. If you want it to try and figure it out I would send it to you. 
I would love to get these thermostats working on our baseboard heaters.

klausahrenberg, 2019-11-19:
Did yo ever received a correct time via MQTT? That means, the device shows no or wrong time, but via mqtt you recived a correct local time?

cclein, 2019-11-19:
Not this unit, no. I think it was a dud from the factory. This one unit, with the original firmware would never pair with the app. I video it and sent back to moes house and they sent me a new unit which I could put your firmware on it.

cclein, 2019-11-19:
Is there some where we can communicate a little bit quicker? Discord maybe?

klausahrenberg, 2019-11-19:
But if you received an replacement unit from moes house than it seems, they was sure that you really received a defective unit before. The only thing you can try, is to replace the Tywe3s module with an ESP-12F and hoping that only the Tywe module has a defect. If this doesnt work, just scrap this thermostat and order a new one.

cclein, 2019-11-19:
I replaced the tywe3s modules with the esp-12f It  behaved the same, I refalshed the tywe3s modules with tasmota and it worked fine. That's why I suggested I send it to you???? Otherwise I test it on our Canadian AC240v.

I'm always worried about using european / asian 240v  devices in canada.

bikemike, 2019-11-19:
Maybe the power supply for the wifi module was bad or something to that effect so the it wasn't even booting up.

As for 240v, I have a C17GH3 hooked up and running fine here in Canada. Wish I could find a good adapter for the junction box though. I just cut a hole in a blank oversized face plate and that seems a bit sketchy. I will be trying out my MoesHosue BHT-002-GBLW thermostat tomorrow.

closed 2019-12-11 by klausahrenberg

diabl0w, 2020-02-09:
sorry to resurrect, but were you not getting the device to get into AP mode (wifi icon blinking by holding down a button according to the instructions)? Thats the issue I am having, I measured the voltage going to the wifi module and it was only 0.8V, from the spec sheet it looks like it requires 3.3 volts, so they must have made the PCB wrong connecting it to the wrong trace. I can probe to try and pull 3.3v from another place on the board, it looks like this is what you did? What would be the best place to pull the voltage from, one of the voltage regulator IC's?