klausahrenberg/WThermostatBeca #29: AP mode ok, can't connect to wifi

shin-01, 2019-10-26:
I've flashed my decdeal (Beca) BT-6000-GCLW,  Flash went smooth.
After flashing, I press down button for 8 sec. wifi start blinking, AP is on,
in configuration I put my SSID and psw. 
It reboots, no connection to my network. 
I've tried this at least 10 times. 
any hint?

thank you

ahmaddxb, 2019-11-03:
I think you might be having problems because your SSID has both 2.4Ghz and 5Ghx bands on it try separating your SSID bands on your router to SSID 2.4G and SSID 5G. Then try connect to SSID 2.4G and see if it work. The ESP chip can only connect to 2.4Ghz Wifi Band.

shin-01, 2019-11-03:
yes, i confirm this was my problem, after separating the 2 network, connection is fine.
I'm currently struggling over mqtt, but that's another story. Thank you  very much

closed 2019-11-03 by shin-01

strizl, 2019-12-09:
Hi!
I have the same issue with a BAC-2000 and a BECA_WIFI_HF_V230V2 (should be similar to BHT-1000) device.
ap mode --> wifi configured --> reboot --> no connection to WiFi.
My network is only 2.4G and has the name "Pretty Fly for a WiFi".
I will retry with another WiFi network. Is there a restriction to name/length/no spaces, etc.?

In addition, it seems that I am not able to restart AP mode by holding temp.-down-button again --> wifi symbol is blinking, but no AP is created by thermostat where i can connect to.

Any idea is very welcome!
Thanks a lot!