klausahrenberg/WThermostatBeca #32: How can i turn on manualMode & Lock thermostat

rengaboy, 2019-11-03:
Hello everyone.
I issued a problem, when i switch thermostat manually to scheduled mode, i can't switch it back with mqtt publish command. I also dont't understand, wnat topic i have to use with ecoMode, locked values. When i try to publish {\"manualMode\":true} to topic thermostat/hall/state, nothing happens.

rengaboy, 2019-11-03:
i found mcu value already "55 aa 01 07 00 05 04 04 00 01 00", but when i try to publish it to mcucommand topic, also nothing happens

ahmaddxb, 2019-11-03:
its thermostat/hall/manualMode and in the payload its true or false

This is an example of mine.
[image]

rengaboy, 2019-11-03:
Thank you, @ahmaddxb, i just found solution, after 10-time careful reading of readme.

closed 2019-11-03 by rengaboy