klausahrenberg/WThermostatBeca #33: BHT-002-GCLW WiFi Symbol

31warren, 2019-11-05:
Great software - all functions working well on two BHT-002-GCLW devices.

One minor issue is that the WiFi symbol doesn't show up in the display when the device is connected to WiFi.

Any suggestions how to get this working?

Thanks.

bikemike, 2019-11-09:
As with the original firmware, the icon is only shown during wifi config. There is no way to show it during normal operation.

klausahrenberg, 2019-11-15:
Answer of bikemike is correct. There is no possibilty to switch the wifi symbol on or off so far. This firmware controls only the ESP inside the thermostat. The thermostat itself has an own controller. From ESP side I can only use the existing commands. That's why not possible.

closed 2019-11-15 by klausahrenberg