klausahrenberg/WThermostatBeca #34: TimeZone from Fitzbox

JoFrGa, 2019-11-06:
Hello,
i'm trying to use my fritzbox as an ntpserver. For this purpose i stopped in the fritzbox the connection to the internet.
The result is, that the shown time on the thermostat is the Z-time instead of the normal CET (17:00 inste of 18:00).
I think, this is an result of the Loop-part of KaClock.cpp (after 2.Sync of time), but i don't know how to change the program.
thank you.
Jochen

klausahrenberg, 2019-11-15:
The time sync has 2 components. From NTP server the thermostat can only request the Z-time that's correct. The second component is that the firmware tried to evaluate the actual time zone based on IP address via the great api 'http://worldtimeapi.org/api/ip' in KaClock.cpp. Therfore the thermostat needs an access to internet. A NTP server does not have any timezone functionality.

klausahrenberg, 2019-11-15:
The only idea is that your Fritzbox can convert the received NTP time to your local time and provide this modified time as a NTP time to your clients. Than the thermostat has no problem to deal with that...

closed 2019-11-15 by klausahrenberg