klausahrenberg/WThermostatBeca #37: WiFi Reconnect Issues

coldfire84, 2019-11-14:
I've had to power of/ on my thermostat a couple of times in the last 3 months as it has dropped off the network and never reconnects (MQTT messages stop working/ there is no assigned DHCP lease etc). I suspect my automated AP firmware update causes the wireless to disconnect, but the thermostat simply never reconnects and is the only IOT device (of about 30) that doesn't. No SoftAP etc, it's just offline.

The thermostat itself works, in that the relay is triggered at target temperature and the target temperature can be adjusted.

Not sure if it matters but I tend to run with webServer set to false.

Is there any way to increase logging/ send serial logs to MQTT or another source? 

Has anyone else experience issues such as this?

cclein, 2019-11-14:
This just happened to me two days ago.

klausahrenberg, 2019-11-15:
Yes it's correct. Reconnection doesnt work well until yo restart the device - which is quite difficult by a wall-mounted device :-) I will fix it in a future version.

klausahrenberg, 2019-12-11:
Could you please install version 0.97 and test again? Reconnection should be possible now.

klausahrenberg, 2019-12-12:
New version 0.98 with several bug fixes. Please take attention to following: 
- After update all settings must be reconfigured
- MQTT topics has changed completely between v0.95 and later, see Readme.md for new json structure

closed 2019-12-15 by klausahrenberg

iMike78, 2019-12-15:
I have two BHT-002-GCLW. One of them has the 99a the other one is the 99b FW on it. Both of the are loose the wifi connection within 15 minutes after I restarted them, and never reconnect. The older one had the 95 on it before, there was never problem with it. Will I need to downgrade to have a stable wifi connection, or is there any solution for this? Cheers

klausahrenberg, 2019-12-15:
Try 0.99c. Connection was lost and did not reconnect. Hope, it's solved, I'm on testing now and it looks good so far. This issue is tracked here: https://github.com/klausahrenberg/WThermostatBeca/issues/57