klausahrenberg/WThermostatBeca #39: Work on Tasmosta

emanuelecavestri, 2019-11-20:
Hi,
you did a fantastic job but in my opinion I think you can start develop with the tasmota firmware.
Tasmota already has many features developed and are now implementing communication with tuya MCU

ivanmarban, 2019-11-22:
The big question here is if it is possible to control the relay status using Tuya Serial Port Communication Protocol rather than making those hardware modifications that @bikemike did. Maybe @klausahrenberg could know something about it.

klausahrenberg, 2019-11-22:
Short answer: No, Relay status is definitely not reported/controllable via Tyua Serial Port Communication. You have to modify the hardware, but it's an easy think.

klausahrenberg, 2019-11-24:
Tasmota is great - no question. However I want to have Mozilla Webthing support for my thermostat. That's why I not switch to Tasmota, instead the next version will include this support. Everyone can build the support in Tasmota, all necessary MCU commands are inside my code and explained there.

closed 2019-11-24 by klausahrenberg