klausahrenberg/WThermostatBeca #42: Boot mode

attika87, 2019-11-29:
Hi,
I am blocked at the very beginning. I did the cabling to do the flashing, but looks like cannot put it into boot mode. Do you have any recommendation what i can do? I followed everything but cannot upload the new firmware. 

Thank you for your help!
Attila

klausahrenberg, 2019-12-03:
After cabling you have to bring the unit in "flash mode", not boot mode. For this you have to connect GPIO0 to GND at powering the ESP module with 3.3v. So follwing order:
- Don't connect the thermostat to AC power, all flashing steps should be made only via programer voltage
- Wire the RX-connection of ESP to TX of programer, and TX of ESP to RX of programer
- connect GND and GPIO0 of ESP to GND of programer, and VCC of ESP to 3.3V of programer
- connect USB cable to programer, after power up you can disconnect GPIO0 from GND - unit is now in 'flash mode'
- Now you should be able to make an backup of exisiting firmware according to my Readme
- After backup the unit ist in 'boot mode' again, so to bring it again in 'flash mode', disconnect the USB cable and connect GPIO0 to GND and connect USB cable again.

closed 2019-12-11 by klausahrenberg

realthk, 2019-12-11:
Perhaps it might be related to issue #5 and the problem is caused by plugging in the programmer to an USB port with the thermostat already connected to it.
For some strange reason, I could not reflash this device after the first time, no matter how I tried.
The solution was to first plug the programmer to an USB port, and connect its 3.3V wire only later to the thermostat (and of course, with GPIO0 to GND)

attika87, 2019-12-15:
It is a magic :) I was able to flash the device. Thank you for your help guys!