klausahrenberg/WThermostatBeca #44: Tasmota 7

baha00, 2019-12-03:
Not an issue just warning, do no load this firmware if you have tasmota v7 already loaded...

cclein, 2019-12-03:
What happens?

klausahrenberg, 2019-12-03:
Please explain the issue, I can't see any problem. Actually you can load my firmware via Upload function of Tasmota and you can go back to Tasmota via Upload function of my firmware. I tested this last week with latest Tasmota 7 nightly/developer without any issue.

cclein, 2019-12-04:
Using the latest tuya-convert I'm pretty sure I ended up with version 7 on my thermostat before loading this firmware. I experienced no ill effects. What should I be looking out for?

baha00, 2019-12-07:
HI all update. Thermostat entered in lock mode. holding up&down arrow turns off lock mode and power&down arrow activates hotspot. No issue here just stupid user= ME

closed 2019-12-07 by baha00