klausahrenberg/WThermostatBeca #45: Flash possible with tuya-convert

helmar74, 2019-12-04:
Just for you information:
I was able to flash my thermostat with the latest version of tuya convert:
([https://github.com/ct-Open-Source/tuya-convert](https://github.com/ct-Open-Source/tuya-convert))

You don't have to solder anything. 
I followed exactly the howto of tuya convert.
Maybe you can add it to your documentation

radokristof, 2019-12-04:
For me it didn't worked on a BHT-6000. Which device did you tried?

helmar74, 2019-12-05:
Hi,
I have a BHT-002-GBLW. Did you tried the latest version of tuya-convert?
In which stage of the flashing did you fail?
(Maybe a outputlog or screenshot could help)

radokristof, 2019-12-05:
Here is the issue regarding this. It looks like the device don't use an IP address from the provided DHCP range. Did you connected the thermostat to the official app before using this converter?
https://github.com/ct-Open-Source/tuya-convert/issues/439

helmar74, 2019-12-05:
OK, no my thermostat wasn't connected either to a WLAN nor to the app.
It was like a virgin, as I flashed the firmware with tuya convert

radokristof, 2019-12-05:
So this is a difference, because I wanted to try out before starting to mess with the firmware...
And now it seems that I can't get the thermostat to delete all config... it always connects with that IP address...

radokristof, 2019-12-05:
I was able to do it with another setup. So it really works, it could be added to the README. I will do it later.

PR #47 (2019-12-11): Added tuya-convert method to README
@klausahrenberg
Fixes #45
closed 2019-12-11 by klausahrenberg