klausahrenberg/WThermostatBeca #46: Can't connect to MQTT server

radokristof, 2019-12-05:
Flashing is done, firmware seems fine. However I have some problems, hope someone can help me:

- After connecting to my network I can see it that it connected to the network, also I can ping it's IP address. However I can't access it on the web server. The web server by default is disabled? Can it be enabled by default? It would make configuration easier.
- MQTT connection is not working... I have set the credentials and the topic, but nothing is received. Also how can I provide the MQTT server? I have tried with <IP-ADDRESS> or <IP-ADDRESS>:<MQTT-PORT>.

What is the right way to do? Also I will propose some README changes which clarifies some things...

klausahrenberg, 2019-12-06:
Hi, webServer is disabled by default. Press the right button on Thermostat when it's off for 8 sec to start the webServer again to change the configuration. At MQTT must be a configuration issue, please check. Once MQTT works, you can enable and disable the webService via MQTT command.
However, next version will enable webServer per default.

klausahrenberg, 2019-12-06:
At your question for payload. '0' should be the payload...

radokristof, 2019-12-06:
Thanks!
I have tried to change the configuration multiple times. Wi-Fi seems ok, because it connects to the network.
However I can't get MQTT to work. Do you use your device by it's hostname and not IP? I have tried passing the IP address of the MQTT broker to the thermostat or IP address + port but neither one worked...

radokristof, 2019-12-06:
Also I have to restart every time the thermostat for the AP to appear. Just turning "off" and pressing the button shows the blinking Wi-Fi but no AP is present.

radokristof, 2019-12-06:
I don't know what happened but:
I have tried connecting the thermostat to my test MQTT server without credentials. It worked instantly. Now I'm back at my normal MQTT server and works there also...

closed 2019-12-06 by radokristof