klausahrenberg/WThermostatBeca #48: Flash Works -> After assigne WiFi -> Bot able to boot

lokidaibel, 2019-12-09:
I used tuya convert to load the firmware. It setup the Wifi Spot.
I can assigned my WiFi but after that the device does nothing and i get a ESPxxxx Wifi shown in the Wifi overview.

If i install tasmota on the device it works.

Its a BECA-6000.

If you want to play around with a Beca 6000 i can donate one. Have some spare ones.

klausahrenberg, 2019-12-11:
Could you please install version 0.97 and test again?

lokidaibel, 2019-12-11:
I successfull installed 0.97 the following way:

I long pressed he down Button to bring my device in the Boot mode. This way he enabled the webserver again and i activated WebThings again.

Now restarted the device (disconnected the display from the wallmount).
After that i was able to patch 0.97 ! Good work so far.

klausahrenberg, 2019-12-12:
New version 0.98 with several bug fixes. Please take attention to following: 
- After update all settings must be reconfigured
- MQTT topics has changed completely between v0.95 and later, see Readme.md for new json structure

closed 2019-12-12 by klausahrenberg