klausahrenberg/WThermostatBeca #49: Flash, AP-mode, settings... OK - but Thermostat does not connect to WiFi

strizl, 2019-12-09:
Hi!

I have the same issue with a BAC-2000 and a BECA_WIFI_HF_V230V2 (should be similar to BHT-1000) device.
ap mode --> wifi configured --> reboot --> no connection to WiFi.
My network is only 2.4G and has the name "Pretty Fly for a WiFi".
I will retry with another WiFi network. Is there a restriction to name/length/no spaces, etc.?

In addition, it seems that I am not able to restart AP mode by holding temp.-down-button again --> wifi symbol is blinking, but no AP is created by thermostat where i can connect to.

Any idea is very welcome!
Thanks a lot!

_Originally posted by @strizl in https://github.com/klausahrenberg/WThermostatBeca/issues/29#issuecomment-563410683_

klausahrenberg, 2019-12-10:
Hi, length of WiFi name is limited to 32. About spaces in name I'm not sure, but shouldn't be a problem. Try version 0.96, I changed a lot at connection behavior.

klausahrenberg, 2019-12-11:
Could you please install version 0.97 and test again.

closed 2019-12-11 by klausahrenberg

strizl, 2019-12-28:
Check with v1.00 and works fine!

Thank you very much!