klausahrenberg/WThermostatBeca #51: Update to 0.96 -> How can I start webserver and communicate with the device

helmar74, 2019-12-10:
Hi,

I've updated succcessfully to 0.96
... but how can I communicate with the device again?
I enabled MQTT, the device sends status to broker.
But I can't send any command.
With version 0.95 this worked:  mosquitto_pub -h localhost -t "thermostat1/state" -m "0"
Now I don't get the status back.
Have the commands/topics changed?

Thanks a lot

klausahrenberg, 2019-12-10:
Yes, It has changed. I will update the doc asap.
To get the state:
Send MQTT topic without any payload to: <mqtt_topic>/things/thermostat/properties

Webserver changed to:
http://<your_ip>/config

Changes were needed to get compatibilty to Webthings.

klausahrenberg, 2019-12-10:
To set properties, send a json with values to <mqtt_topic>/things/thermostat/properties or for one property, e.g., deviceOn to <mqtt_topic>/things/thermostat/properties/deviceOn with a boolean in payload.

helmar74, 2019-12-11:
Thanks a lot for updating the docs.

closed 2019-12-11 by helmar74