klausahrenberg/WThermostatBeca #52: 0.96 - Only RestInterface after update

lokidaibel, 2019-12-10:
hey, i updated the firmware to 0.96

After this i get only the following in the webinterface:

[{"name":"thermostat","href":"/things/thermostat","@context":"https://iot.mozilla.org/schemas","@type":["Thermostat"],"properties":{"temperature":{"title":"Actual","type":"number","readOnly":true,"unit":"celsius","@type":"TemperatureProperty","href":"/things/thermostat/properties/temperature"},"targetTemperature":{"title":"Target","type":"number","unit":"celsius","multipleOf":0.50,"@type":"TargetTemperatureProperty","href":"/things/thermostat/properties/targetTemperature"},"deviceOn":{"title":"Power","type":"boolean","@type":"OnOffProperty","href":"/things/thermostat/properties/deviceOn"},"schedulesMode":{"title":"Schedules","type":"string","enum":["off","auto"],"@type":"ThermostatModeProperty","href":"/things/thermostat/properties/schedulesMode"},"state":{"title":"State","type":"string","readOnly":true,"enum":["off","heating","cooling"],"@type":"HeatingCoolingProperty","href":"/things/thermostat/properties/state"}}}]

What can i do next ?

klausahrenberg, 2019-12-10:
Ok, there are 2 options for what next:
1. If you use webthings, you can add the thermostat to your gateway now. Webinterface > Add device.
2. Use and control the thermostat via MQTT. It has to be configured via Web-Interface of thermostat. To access the web interface again, use path 'http://<your_ip>/config'. The root displays the json above - it's needed for webthings.

I will update documentation today asap.

lokidaibel, 2019-12-10:
Ok works. But there is no option to select the BHT-6000 ?

Greetings

Lukas

klausahrenberg, 2019-12-10:
But 6000 should be identical with BHT-002 floor heating

closed 2019-12-11 by klausahrenberg

lokidaibel, 2019-12-11:
It seam to work. 

Greetings

Lukas