klausahrenberg/WThermostatBeca #54: Firmware Update - 0.96 -> 0.97 Timeout

lokidaibel, 2019-12-11:
Hey,

when i try to update to the new Firmware i get a timout.
After short time i can call <IP>/config again. But seems i stuck with 0.96 for now.

Also i cant change anything under http://<IP>/wifi
as soon i click on save i get a timeout.

Any idee to how update the firmware to 0.97 ?

lokidaibel, 2019-12-11:
After i tryed the update, the System went dark. No more reaction even after resett power...
Be carefull with this FW release.

radokristof, 2019-12-11:
Same for me, it works sometimes, but really-really slow. And usually it just unreachable or can't change anything.
0.96 is same? If yes I will roll back to 0.95

lokidaibel, 2019-12-11:
I am unable to roll back. Stucked at 0.96. Seams he where unable to load the FW.
Also i disabled the webinterface... I thing thats my main issue here, MQTT working. Is there any way to enable Webinterface over MQTT commands ?

After that i will face the issue with the failed FW update.

klausahrenberg, 2019-12-11:
Please wait at the firmware update. you get a timeout because you have to reconfigure the thermostat after update from 0.95 to 0.96. Please look for a WiFi hotspot after update and setup again

klausahrenberg, 2019-12-11:
I know, the update is not working nice yet - because 'sucess' message is missing, but the update itself is working. I will optimize this in next updates...

radokristof, 2019-12-11:
I have tried to update from 0.95 to 0.97. I have tried reconfiguring the Wi-Fi but it didn't helped. Connected to the Wi-Fi but after accessing it from the web, only a JSON is returned no webconfig, also no MQTT, however I also set it up at configuration.

radokristof, 2019-12-11:
Now I'm back at 0.95 which works great...

klausahrenberg, 2019-12-11:
Ok, please don't disable Webthing in configuration, otherwise the webServer is going down and can't be enabled again via MQTT. I will correct this. To enable the webServer again: Hold the Down button for 8 sec to get in Config mode or, if not working, disable your WiFi that the thermostat can't connect - then the AP should come up again.

radokristof, 2019-12-11:
I didn't disable the WebThing, only enabled the MQTT and entered the credentials. What happened after this is it connected to my Wi-Fi but was not accessible through MQTT. The webserver also just returned a JSON object. If I entered pairing mode I was able to see the webserver and my config was there...

klausahrenberg, 2019-12-11:
Then take also attention that mqtt topics and json structure have changed since 0.96, because of webthings compatibility.
See here:
https://github.com/klausahrenberg/WThermostatBeca/blob/master/README.md#1-state-report

radokristof, 2019-12-11:
That makes sense, thanks I will retry then!

Where is the state provided? It says that an empty payload can be made to get the properties, but it will also return that here:
<your_topic>/things/thermostat/properties

klausahrenberg, 2019-12-11:
Yes, send to this topic an empty payload (not with a '0' value like in 0.95) and the device sends to the same topic a json-payload back. Or just change somethin on the thermostat, state will send to mqtt server

radokristof, 2019-12-11:
Thanks, I will retry! However I still don't understand why the webserver remained unavailable and just a JSON can be seen there. This is because Mozilla WebThings?

klausahrenberg, 2019-12-11:
Yes, sorry for the confusion, please use 'IP/config'. Webthings needs the root path to work...

radokristof, 2019-12-11:
Ok thanks now I saw that in the README. And you say that after I upgrade from 0.95 to 0.97 I need to reconfigure the Wi-Fi? For me it seemed that it reconnected itself automatically...

klausahrenberg, 2019-12-11:
Could be, for any reason sometimes it connects even the settings for wifi are lost.

radokristof, 2019-12-11:
I know what happens in the webConfig. I can load on menu but after that any other menu will timeout. I need to wait some minutes and I can load any... Then others become unavailable...

klausahrenberg, 2019-12-12:
New version 0.98. Several bug fixes on configuration. Please take attention to following: 
- After update all settings must be reconfigured
- MQTT topics has changed completely between v0.95 and later, see Readme.md for new json structure

radokristof, 2019-12-12:
@klausahrenberg One problem: I'm on 0.97 and clicking on the 'Update firmware' menu it just loads infinitely. How can I update to 0.98?

klausahrenberg, 2019-12-12:
Strange, I didn't had this issue so far. Try to call firmware page directly: http://your_ip/firmware.
Please also check, if the thermostat did open an WifiAccessPoint, but your thermostat is still connected to your network. I had this strange behavior, when network setup needs to be done.

klausahrenberg, 2019-12-12:
If this doesn't help, disable your WiFi network. The thermostat should open an own Access point. maybe then the configuration works

radokristof, 2019-12-12:
No Wi-Fi AP, calling the firmware page directly also times out. I think creating the AP and updating there will help (this is how I did it before). Hope 0.98 will fix this.

realthk, 2019-12-12:
Same here: trying to upload 0.98 on 0.97 does not work, no response after pressing the upload button. Device is still on the wifi, replies to ping, but code is not running, because no replies to MQTT requests anymore.
After power off-on, it works again - but still on 0.97, so updating probably died before replacing the fw. As it more or less works fine now (with some NodeRed scripts to create standalone topics for Home Assistant instead of a json array) I rather do not flash again by soldering.

radokristof, 2019-12-12:
@klausahrenberg I was able to update to 0.98 by putting it into AP mode and applying the update there.
However I still have some issues:
- Menus can only be opened by double pressing the buttons. When I first press one button it just loads infinitely. If I press again while loading, it will load immediately.
- Wi-Fi AP not visible when pressing the TempDown button. I have to cut power from device and before starting press the TempDown button. This is the only way to have Wi-Fi AP.
- The 'Configure Thermostat' only has BHT-002. Could it be possible to add BHT-6000 there?

realthk, 2019-12-12:
>     * Menus can only be opened by double pressing the buttons. When I first press one button it just loads infinitely. If I press again while loading, it will load immediately.
This seems to be browser-related: I had no problem like that with Firefox, now tried with Chrome, and indeed it happens most of the time.

radokristof, 2019-12-12:
Yes can be the case because I'm using Chrome mostly. Would be great if at least Firefox and Chrome could be supported.

radokristof, 2019-12-12:
Tried it with Firefox. Yes it works much better with Firefox...

klausahrenberg, 2019-12-13:
Ok, I can confirm the problem a upload issue in Chrome, it should be solved in 0.99. There was also a much bigger proplem with endless loops in schedule auto mode, since I implemented that targetTemperature is taken from schedule, when thermostat is in 'auto mode'. This is also solved. However, the webinterface has sometimes some laggy response, so you need to click twice sometimes in Firefox and Google. I will work on this issue.

radokristof, 2019-12-13:
Thanks for confirming those!

radokristof, 2019-12-13:
Also after a day on 0.98 it seems that the thermostat is unreachable on webserver and on MQTT also...

radokristof, 2019-12-13:
After a restart it works as it should. This should be also addressed. Can we somehow fetch some logs from the ESP chip what happens what can cause issues like this? Just to help you mainly...

radokristof, 2019-12-13:
I have updated to 0.99 through Firefox. However I still have loading issues on the webserver in Chrome.

klausahrenberg, 2019-12-14:
Yepp, not working well at the moment. At the moment the firmware freezes after some hours. There is an error implemented by me at the last changes - I'm testing at the moment to find it. I uploaded version 0.99.b. I think, the Chrome issue - I found the reason. My feeling is that it is now normal rsponisive. Freezing issue I'm at testing now...

radokristof, 2019-12-14:
Yes thanks it seems that the Chrome bug is gone with 0.99b.
One thing I have also found: 
- Clicking the reset doesn't actually reset the ESP to defaults?
- Also it would be great if a popup window will appear that you really want to reset, or on a sub-site you should have to press again a button to confirm the reset, just to not reset it accidentially.

klausahrenberg, 2019-12-15:
Reset only restarts the ESP, maybe the wording is wrong.
I will close this topic, freezing (it was more a 'not reconnecting') is solved in 0.99c, but will be tracked here: https://github.com/klausahrenberg/WThermostatBeca/issues/57

closed 2019-12-15 by klausahrenberg

radokristof, 2019-12-15:
Thanks for clarifing it! However then it might be better to be renamed to 'Reboot'?

klausahrenberg, 2019-12-16:
> 
> 
> Thanks for clarifing it! However then it might be better to be renamed to 'Reboot'?

Yes, renamed it to 'Reboot' in v1.00