klausahrenberg/WThermostatBeca #55: Question: Differences between GBLW and GCLW

sven5, 2019-12-11:
Hi,

I accidently ordered the Moes House BHT-002-G**C**LW for gas boilers.
Does anybody know what's the difference to the G**B**LW version? 
I already changed the soldering points from dry contact to wet contact to have it switching the water floor heating actuator. 
However, I wonder if there are software differences between these two?
I'd like to keep the original software because I'm satisfied with it.

Thanks
Sven

klausahrenberg, 2019-12-13:
According to Amazon, the models are different in following way:
BHT-002-GBLW is for floor heating
BHT-002-GCLW is for Gas Boiler
I expect that the Thermostat MCU will make a difference in heating times. So the regulation (switch times) of the relay will be different maybe. However, for this firmware here it makes no difference because it's not controlling the relay.

closed 2019-12-13 by klausahrenberg

sven5, 2019-12-19:
Just for reference. I got a reply from moeshouse.com.
There are no differences in the software. Just the hardware contacts are different.

> 
> Dear Sven,
> No,there are no differences of the two.Thanks for your email.
> 
> Best regards.
> Yours sincerely
> Zoe
> 
> 
> Zoe  (smartlife1@qq.com)    
> 
> My Manager : Amin Lee  (amin@morning-e.com)
> 
> --------------Original Message-------------
> From：....
> To：info@moeshouse.com
> Cc：
> Subject：AW: New customer message on December 19, 2019 at 3:55 am
> Date：2019-12-19 16:08:23.0
> 
> 
> Hi Zoe,
>  
> thanks for your reply. I know that there are differences in hardware layout. But could you please elaborate if there are also differences in the internal software of the MCU of the thermostat? Or are they the same? 
>  
> Thank you very much!
> Sven