klausahrenberg/WThermostatBeca #57: 99b is still hanging

cclein, 2019-12-14:
I just flashed 99b onto a new thermostat. It's Saturday (day 6). ESP is still hanging after a while and unavailable to ping. No schedules are being transmitted.

realthk, 2019-12-14:
Probably affects 0.98 and 0.99 as 0.97 seems to run fine for a few days now.

iMike78, 2019-12-14:
0.99a is running fine on my moeshouse thermostat.

klausahrenberg, 2019-12-15:
Ok, I think, it's fixed now in 0.99c. It's running whole day now without problems. I still leave the topic open and will still test it. But please use 0.99c or later.

radokristof, 2019-12-15:
I have updated to 0.99c. I'm testing now.

cclein, 2019-12-15:
Am testing 99c as well.

klausahrenberg, 2019-12-16:
Ok, im courageous now and updated the firmware to version 1.00. I tested this issue for over 1 day now without problems. From 0.99c above the thermostat should not hang anymore.

radokristof, 2019-12-16:
Yes 0.99c seemed fine for me. Testing 1.0 now, but it looks great. 
One minor thing: In Chrome the first command (e.g.: pressing one of the menu buttons) is still just loads indefinitely, pressing again will load immediately. After these first commands succeeds it works as it should for some time, but after a longer period, it still just shows loading for the first time.

julezman, 2019-12-17:
@klausahrenberg - First of all, thanks for this firmware.  I've been using v0.95 for a few weeks successfully on BHT-3000 GCLW devices.  Secondly, I can confirm 1.00 has been working well for the last 12 hours for me.  Will update if any issues.  0.98 had lots of issues and amazed you've fixed so quickly.  I hope to contribute in time.  Thanks again!

lokidaibel, 2019-12-17:
I can confirm that. two devices running smooth for over 48H will add 4 more (BHT-6000) today and report if they runsmooth aswell.

lokidaibel, 2019-12-18:
I thing the issue is resolved. Stable since 3 days.

klausahrenberg, 2019-12-18:
Ok, thanks for feedback. Also my device is stable now. Issue will be closed now.

closed 2019-12-18 by klausahrenberg

lokidaibel, 2019-12-18:
Sorry to disapoint. After 4 days my first patched lost connection...
Cant access the webpage anymore. Still getting MQTT Data.