klausahrenberg/WThermostatBeca #59: MQTT Documentation - /info Page improvment

lokidaibel, 2019-12-16:
Hey,

the temperature Values work well. Now i want Buttons in my Openhab2 to control this.
I thing we should list our MQTT Topics and maybe also all Webthing informatione on the /Info Page ?

Also if you have, can you print your Openhab2 Item definition / Json Rule definition (or some1 else who has it in place all ready) ?

ahmaddxb, 2019-12-16:
You can have a look at my Openhab files to get an idea how i managed it. You don't have to go into that much detail as I have and you can ignore the schedule part as that needs a node red flow to function.

[Openhab Beca.zip](https://github.com/klausahrenberg/WThermostatBeca/files/3967551/Openhab.Beca.zip)

radokristof, 2019-12-16:
I'm using MQTT2 binding, so I have different schemas. But with that binding, anything is possible just using the binding configuration, no additional transform or anything needed. Just specifiy the incoming value transformation as a JSONPATH transform, and for items which needs to be controlled, like targetTemp, you should also specify the outgoing value transformation as a json object.

lokidaibel, 2019-12-16:
I also use MQTT2. Can you please show me your transformations ?

lokidaibel, 2019-12-16:
@ahmaddxb Thx. exactly what i needed.

radokristof, 2019-12-16:
Because the transformation is not correct.
Set the topic to device/things/thermostat/properties
And the transformation to JSONPATH:$.temperature

ahmaddxb, 2019-12-16:
> 
> 
> @ahmaddxb Thx. exactly what i needed.

Bare in mind I'm still on firmware v0.95 and still using MQTT v1 in openhab so its different for v2.

radokristof, 2019-12-16:
However I agree with @lokidaibel if some info could be shown on the web interface it would be great.
I mean currentTemp and targetTemp maybe. Like Tasmota does. It could help with debugging, etc...

lokidaibel, 2019-12-16:
@radokristof Thats exact what i tried so if my beca has the following name: 
beca_thermo_buero

and i setup my Channel to this, it should work ? 
https://gyazo.com/914ca94df8637a89f953f7d1d5dae645

I setup my mosquitto right as i can see the stats beeing procede but still i cant se any Value in my Interface.

{"idx":"beca_thermo_buero","ip":"192.168.0.97","firmware":"1.00","temperature":24.50,"targetTemperature":24.00,"deviceOn":true,"schedulesMode":"off","ecoMode":false,"locked":false,"floorTemperature":0.00,"state":"heating"}

So any suggestions here ?

I setup everything with PaperUI and Mqtt2.

radokristof, 2019-12-16:
Yes this should work and works as you just wrote.
Use the same method for getting the targetTemperature state.
MQTT state topic is the same, set MQTT command topic to the same topic as the state topic.
So both of them is thermostat/things/thermostat/properties.

set incoming value transform to JSONPATH:$.targetTemperature. And the output value transform to { "targetTemperature": %s }

attika87, 2019-12-18:
Help me pls. 
I have OH2 mqtt1. 
I see the mqtt topic, but cannot read anything from there. I have an item like that:
Number DolgozoPadloTemp {mqtt="<[mqtt:/dolgozo/things/thermostat/properties:state:JSONPATH($.floorTemperature)]"}
Any idea? 
jsonpath transformation is istalled but i did nothing else? Some configuration missing?

Himbi777, 2019-12-18:
Hello 
I also have a problem:
how to send deviceOn or send the targetTemeprature to the BHT002 ?
I tried several options but none worked
my Topic is   /Beca/BHT01/things/clock/properties
Publish 1 :  /Beca/BHT01/things/thermostat/properties/deviceOn false   --> no function
Publish 2 : /Beca/BHT01/things/thermostat/properties {"targetTemperature": 21.0}  --> no function
Publish 3 : /Beca/BHT01/things/thermostat/properties/targetTemperature 27  --> no function
Publish 4 : /Beca/BHT01/things/thermostat/properties {"deviceOn":false}        --> no function

where is my mistake? Any idea ?
I used MQTTfx for testing

radokristof, 2019-12-18:
What is the topic set? Usually MQTT topics start without a "/". 
Publish 4 : /Beca/BHT01/things/thermostat/properties {"deviceOn":false} --> no function

This should work. Try setting some simpler MQTT topics first in the device setting, like "thermostat".

Then publish here: thermostat/things/thermostat/properties {"deviceOn":false}

radokristof, 2019-12-18:
@attika87 Sorry I can't help with MQTT1. I have changed all of my MQTT Things to v2 over a year ago...

attika87, 2019-12-18:
> @attika87 Sorry I can't help with MQTT1. I have changed all of my MQTT Things to v2 over a year ago...
> But you also seem to start your topic with a "/" character. This how you set up the topic in the device? This is not usual...

sh..... the "/" was the problem. Thank you!

strizl, 2019-12-27:
> Yes this should work and works as you just wrote.
> Use the same method for getting the targetTemperature state.
> MQTT state topic is the same, set MQTT command topic to the same topic as the state [...]

@radokristof 
Can you pls help me with MQTT output syntax - I'm struggling with it.
Regarding the information you already gave, I have the following openhab item:

`Number Thermostat1TempSoll "Soll-Temperatur Wohnzimmer [JSONPATH($.targetTemperature):%s ]" { mqtt=">[openhabbroker:thermostat1/things/thermostat/properties/:command:command:*],<[openhabbroker:thermostat1/things/thermostat/properties/:state:JSONPATH($.targetTemperature)" }`

The input works fine - but output creates errors.
Do you have a hint for me?

Thank you very much!!

radokristof, 2019-12-27:
@strizl Sorry I don't use the MQTT1 binding anymore and I can't really remember what was supported there and how...
Just a few things I found looking at your definiton:
- Do you really need the two JSONPATH transformation for the inbound state? If the MQTT binding can parse the JSON then you don't need to specify the item label again with the JSONPATH transformation.
And I think this is where this config is wrong:
- The JSONPATH inbound is not working correctly, just passes the whole payload ( I mean this part: <[openhabbroker:thermostat1/things/thermostat/properties/:state:JSONPATH($.targetTemperature)" is not correct), so you apply the transformation to the label.
However this transformation is only applied to the visible label, not for the state, but I might be wrong. Try removing that form the label, have it like this: "Soll-Temperatur Wohnzimmer [%s]". If it works then ignore this.
Secondly on the inbound you need to send also a JSON object. Now you just send a simple number back to the broker. I might be wrong but something like this should work:
>[openhabbroker:thermostat1/things/thermostat/properties:command:*:{ "targetTemperature": ${command} }]

strizl, 2019-12-28:
@radokristof 
**Thanks a lot for your fast response!
You gave the hint to a working solution.**

If others have the same problem, here's the solution for previous MQTT:
`Number Thermostat1TempSoll "Soll-Temperatur Wohnzimmer [%.1f °C]" { mqtt=">[openhabbroker:thermostat1/things/thermostat/properties/targetTemperature:command:*:default],<[openhabbroker:thermostat1/things/thermostat/properties:state:JSONPATH($.targetTemperature)" }`

attika87, 2020-01-03:
> @radokristof
> **Thanks a lot for your fast response! You gave the hint to a working solution.**
>
> If others have the same problem, here's the solution for previous MQTT:
> `Number Thermostat1TempSoll [...]

i struggling with the mqtt controll (incomming mesasges are working fine). I wan to set the targettemperature only but no success. I have only this item and a rule with postUpdate. 
Number DolgozoTargetSend {mqtt=">[mqtt:dolgozo/things/thermostat/properties/targetTemperature:command:*:default]"}
This json thing is driving me crazy :)

radokristof, 2020-01-04:
Why don't you use the MQTT2 binding?
You can use both binding simultaneously, so you can convert all your MQTT items gradually.
The MQTT1 will become completely obsolete in this year when OH3 will be out which will remove the 1.X compability layer and also that binding is not maintained anymore.
With the MQTT2 binding you can easily specify the outgoing value transformation (only one line in the binding setup).
If you want to stick with MQTT1, then you can only do that in a rule.
Show me the rule you created for sending the values to the Thermostat. 

> I have only this item and a rule with postUpdate.

If you use postUpdate in that rule where you want to send the value to the Thermostat, this can be the first problem. postUpdate won't send the data to the Thermostat, just update the Item in openHAB.

closed 2020-04-03 by klausahrenberg