klausahrenberg/WThermostatBeca #61: Temperature configuration in Manual Mode has a 0,5 °C glitch

lokidaibel, 2019-12-18:
So if i configure 24 °C in manual mode and the IR Sensor get 23,5 °C he turn off the heating. 
I guese that has something to do with the /0,5 caclulation you doing when transforming the MCU values ?!

Anyone else can confirm this ?

radokristof, 2019-12-18:
I think this is not related to the firmware. It will do this even with the original firmware, because the ESP chip only controls communication mainly, so how the relay works (when it turns on/off) is not related to the firmware.
And I think the problem is that these devices hysteresis is limited to 1°C (in the manual it mentions as 'Deadzone Temperature' and you can't set it lower than 1°C). So that means it can't switch ON/OFF the thermostat when the difference between the actual and target temperature is less than 1°C.

realthk, 2019-12-18:
I quite often set "20,5" and it works fine: thermostat starts to heat at 19,5 and stops when reaches 20,5.
Now accuracy of this is another matter (my Xiaomi and BME280 sensors shows different value than this thermostat), so temperature correction (and hysteresis) less than 1C would be useful, but you can only set these in 1 degree steps on this model. Probably it's because this thermostat has far less accurate sensor, so it won't work correctly anyway.

Perhaps the best solution would be if we were able to control the relay, not only get the status of it, and creating a better thermostat with better temperature sensor and software.

klausahrenberg, 2019-12-18:
Hi, as @realthk already mentioned, this behavior is out of the scope of my firmware. But anyhow, it's correct that the thermostat switches off the relay before the targetTemperature is reached. The thermostat calculates with an 'overswing', that means, the heater is still hot for a while after switching off the relay and will heat longer. If the thermostat would switch the relay off just if targetTemperature is fine, the resulting room temperature would be too high.

closed 2019-12-18 by klausahrenberg

realthk, 2019-12-18:
Yes, better thermostats do that, but have you experienced it on this simple model as well?
My BHT-002 gas heater version seems to switch off always right at the targetTemperature (so a few tenths of overswing can be measured on a sensor with better, 0,1C resolution) and switches back on 1C below.

I think this is still somewhat better than on some other cheap chinese thermostats which place targetTemp in the middle and switch on 0,5C below/above it.

klausahrenberg, 2019-12-18:
Yes, maybe you are right, these cheap thermostats maybe switch only the relay 0.5C before. Better thermostats should calculate the relay time depending on how big the difference between actual and target temperature is.