klausahrenberg/WThermostatBeca #62: Manual temperature change doesn't send MQTT message

realthk, 2019-12-24:
When targetTemperature is changed on the device with the physical buttons, there is no MQTT message created about it, unless it resulted state change.

It is not a huge problem because
-  if manual temp change was high enough for the thermostat to switch on the relay there is a "properties" message sent
- after every 5 minutes or so "properties" message is sent anyway
but it means that thermostat display in your home automation software differs from the real device for a few minutes, so it would be nice to have MQTT message generated every time when targetTemp is changed on the device - if possible of course (that is the MCU sends info to our wifi chip at once)

strizl, 2019-12-28:
For me the same - would be a nice feature! But thx for the improvements and support so far!!

austwhite, 2020-01-12:
Adding my name to the nice to have this.
Also, I noted in Version 1 that all MQTT messages are sent twice when they send.  Time, status etc all get sent twice.

sylvaing, 2020-01-27:
adding my name also ! :-)

Bettman66, 2020-03-08:
I have compile the firmware 1.0 with 10sec-intervall.
[https://github.com/Bettman66/ThermostatBecaWifi/blob/master/firmware1.0_10sec.bin](url)

nimda5, 2020-03-10:
You have great platformio config too, thanks.
Changes your done in `stateNotifyInterval`?
Can you move it to config and maybe you will see problem like other see, where after some time only date is sent from device to mqtt.
 
> I have compile the firmware 1.0 with 10sec-intervall.

Bettman66, 2020-03-11:
The platforimio.ini is for the old firmware.
For the new firmware 
`[platformio]
src_dir = 
build_dir = .pioenvs
build_cache_dir = .cache

default_envs = beca

[esp82xx_defaults]
build_flags               = -D NDEBUG
                            -mtarget-align
                            -Wl,-Map,firmware.map
                            -Wno-deprecated-declarations
                            -Wp,-w

[common]
framework                 = arduino
board                     = esp01_1m
board_build.flash_mode    = dout
platform = espressif8266

[env:beca]
platform = ${common.platform}
framework = ${common.framework}
board = ${common.board}
lib_deps =
  NTPClient
  Time
  DNSServer
  EEPROM
  ESP8266HTTPClient
  ESP8266mDNS
  ESP8266WebServer
  ESP8266WiFi
  Hash
board_build.ldscript = eagle.flash.1m.ld
build_flags = ${esp82xx_defaults.build_flags}`

Yes, i have change stateNotifyInterval in WDevice.h.

nimda5, 2020-03-11:
Yes, thanks, I found that reading original 1.0 requirements.
I added https://github.com/klausahrenberg/WAdapter.git to the lib_deps. Bus to change  stateNotifyInterval  requires little different includes... But it's not hard.

fashberg, 2020-03-17:
Hi
using my fork  this is fixed.
Sending the Update was commented out in the code.

You have to change MQTT Topics, it's now devided between cmnd and stat (see https://github.com/fashberg/WThermostatBeca#integration-in-home-assisant )

Check out https://github.com/fashberg/WThermostatBeca  (also you need my version of  WAadpter, or just try binary 1.02b)

No response from klausahrenberg on my pull request.

realthk, 2020-03-17:
Great! 
Not sure now, it was long ago in december when last tested it, but I fear HA doesn't handle really well when more values parsed from the same JSON array. it seemed like it was caching values and detect change only in a single field even when more fields were changed in the same message. 
That's why I [started to use a Node-red flow](https://github.com/klausahrenberg/WThermostatBeca/issues/43#issuecomment-565699253) to extract all fields into their separate mqtt topics.
Anyway, even if that's still the case (but HA might have changed in the meantime) the important part is your fix about reporting the manual temp change at once, thanks! I'll try it soon.

klausahrenberg, 2020-04-03:
> When targetTemperature is changed on the device with the physical buttons, there is no MQTT message created about it in version 1.0, unless it resulted state change.

It's fixed in version 1.02

closed 2020-04-03 by klausahrenberg