klausahrenberg/WThermostatBeca #63: BHT 002 gets very hot and shows a floor temperature even though no sensor is connected.

Himbi777, 2019-12-31:
Hi there
I flashed a total of 4 BHT-GCLW with the firmware. Two BHT are going great, the other two are getting very hot. Instead of "Err" they show a floor temperature. Likewise, these can no longer be flashed via the serial interface.
Does anyone know of this problem - and how can it be fixed?

austwhite, 2020-01-09:
Hate to say it, but it sounds like faulty hardware.  Usually when the devices get very hot it indicates excessive current draw and that just causes weird errors.   Second thing is you can no longer flash the devices.  That usually indicates the chip is faulty.
Happy to hear other advice, but that is just my experience with these things.

Himbi777, 2020-01-12:
I think I can still use the two devices as paperweights, thanks for the support.
Two new devices have been ordered

closed 2020-01-12 by Himbi777