klausahrenberg/WThermostatBeca #65: Heating and Cooling with the same relay

ducklingasa, 2020-01-04:
Hi,

In my flat, the same two wires control the heating (during winter) and the cooling (during summer).

I'm wondering if there is a way through this firmware to change whether the relay should be closed below (heating) or above (cooling) the defined target temperature.

Thanks for the help!

austwhite, 2020-01-09:
I'll watch this too, but I doubt there is a way without re-writing it every season change as heating works the opposite to cooling.

realthk, 2020-01-09:
I doubt it mostly because this firmware has nothing to do with **controlling** the relay. That is handled by the non-replaceable firmware of another chip, the MCU.

austwhite, 2020-01-13:
I looked into this further.
It is not possible to do that with the single relay thermostats.  These are designed as heating only.
You would need to get a proper heating and cooling thermostat. 
Most HVAC systems cannot operate heat and cool from a single switch as you need to switch heat and cool separately.  
Sorry for the bad news on this.

ducklingasa, 2020-01-13:
Hi @austwhite,

I appreciate your time and effort. I see that I need to explain the situation a bit more, as it's kind of a special and probably not too common system.

The flat is part of a whole house, which has its own heating/cooling solution using water circulating in the walls. The system knows whether it's winter or summer so it changes the water temperate to hot or cold accordingly.

The current thermostat in the flat is only switching the same valve on or off based on the desired temperature. It doesn't know about the rest of the system, but it has a switch which can be turned to heating or cooling (manually) and as I wrote originally it changes whether the relay is closed above or below the defined temperature.

I was trying to figure out if this can be changed or it is only controlled by the MCU logic. This would be the ideal solution.

As a workaround, I can set the temperate to the max during summer which will have permanent cooling (which does not go below normal room temperature as this is not air conditioning) or turn to the min to switch off cooling. It also fine just less ideal.

Thanks!

austwhite, 2020-01-13:
Unfortunately these thermostats would only control heating.  They don't have logic to switch between heat and cool.  They are just a simple on when temp falls below set temp and off when it is above set temp, with some logic for timers.
You would probably be better building a DIY unit using an ESP8266 and a relay module and a summer/winter switch for what you want to achieve.

austwhite, 2020-01-13:
In that case, maybe the two pipe version of the BAC-002-ALW would do the job, but just bypass the fan control component (unless you have multi-speed fan coils).

ducklingasa, 2020-01-13:
Thanks for the suggestions!

closed 2020-01-13 by ducklingasa

lipiss, 2023-02-22:
I vote for this feature, heating and cooling modes. The heat pumps with underfloor heating and cooling are quite frequent in Spain.
Commercial Thermostats have the option to invert the output for cooling. 
To summarize:
1. In general for heating almost all thermoestats use a Normaly Open mode.
2. However for cooling the mode should be Normaly Closed
Lets say the target temperature is 20 degrees and the current temperature is 25, the output would be closing the relay hence activating the valve.
This behaviour could be implemented theoritically at the code?

greatfireball, 2023-02-22:
I am voting for this feature also based on the same situation in our house in Germany as @lipiss described.

lipiss, 2023-02-23:
I see at the code for BAC-002 something about cooling mode:
https://github.com/klausahrenberg/WThermostatBeca/blob/80f9c31756c179941518b3af5650904ce40e52f7/src/WThermostat_BAC_002_ALW.h#L8
Also at the readme it mentions "Only BAC-002-ALW: fanSpeed:auto|low|medium|high; systemMode:cooling|heating|ventilation"
So not really sure if the before mentioned feature already exists.