klausahrenberg/WThermostatBeca #68: MQTT stops responding after a few hours

austwhite, 2020-01-14:
I am using version 1 of the code and the dry contact version of the 002.
It all works well for a time, but then MQTT stops responding.  This can be between 6 and 18 hours, but it never lasts a full day.
Restarting the MQTT server to reinitiate the connection clears the fault.
Restarting the thermostat also clears the issue.
I am using Mosquitto plugin in Hass.IO (latest version).
It seems to be only the MQTT.  WiFi stays connected and the web interface is always available.
Any ideas what might be causing this?  All my other MQTT devices are stable, so I know it is not the MQTT server.  Restarting the server was just a quick way to get the thermostat to reinitiate the MQTT connection for testing, which was proof the code didn't actually freeze.
Thanks

austwhite, 2020-01-14:
Just as an additional note.
I noticed the time status still sends via MQTT but the operational status is stopping, preventing updates from being received.
I also noticed that I can actually change the thermostat setting, but as the status is not returned, Home Assistant doesn't know the new status.

JFingerle, 2020-03-10:
I can confirm that the thermostat stops sending status updates (temperature etc.) via MQTT after a few hours. When this happens the webinterface is still reachable.

austwhite, 2020-03-10:
Just as an additional note.
The problem can be made to occur a lot quicker if the thermostat is changed manually (on the thermostat itself) a few that Mrs in quick succession.

klausahrenberg, 2020-04-03:
I can not reproduce the problem. Please try version 1.02. I integrated a 'running since' in info page. Please check, if the thermostat is running all the time - maybe it crashes after some time. Version 1.02 I have tested for over 12 hours now and it is not crashed and do still send all MQTT messages.

closed 2020-04-03 by klausahrenberg