klausahrenberg/WThermostatBeca #70: Esptool warning "doesn't look like an image file"

mf01, 2020-01-29:
I've just flashed the ESP of my BHTW002GA

`esptool.py -p /dev/ttyUSB0 write_flash -fs 1MB 0x0 WThermostat_1.00.bin`

What's about the "Warning"? Can I ignore this?

```
esptool.py v2.8-dev
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: bc:dd:c2:**:**:**
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Warning: Image file at 0x0 doesn't look like an image file, so not changing any flash settings.
Compressed 65008 bytes to 14232...
Wrote 65008 bytes (14232 compressed) at 0x00000000 in 1.3 seconds (effective 415.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

mf01, 2020-01-31:
Using the backup of the original firmware I don't get this message `Warning: Image file at 0x0 doesn't look like an image file, so not changing any flash settings.`

austwhite, 2020-02-01:
That looks more to do with default settings.
If the firmware works fine you can safely ignore the error.

mf01, 2020-02-01:
Using ESP-power the display is not working. 
So I can close the case and put the thermostat in use?

austwhite, 2020-02-01:
Well if the display isn't working, then it's not working correctly 🙂.
I haven't been able to replicate the issue on version 1.00 or version 0.95 bin files using esptool.

mf01, 2020-02-02:
> Well if the display isn't working, then it's not working correctly slightly_smiling_face.
It wasn't working before, too. I think it's why I only used 3.3V pins from ESP to get and put firmware. I will put all components together and try to power on.

austwhite, 2020-02-02:
Ahh.  Well, if the thermostat has a fault that could be why you are getting the error.
I hope it works for you.

mf01, 2020-02-02:
Ok. Display is working with "real" power. I think fw1.00 is working. because the beca wifi icon isn't blinking for first WiFi setup. But how can I put Wifi data to the device?

austwhite, 2020-02-02:
If the firmware is working it will generate its own WiFi AP.
All the instructions are in the readme.md and installation documentation.
https://github.com/klausahrenberg/WThermostatBeca/blob/master/Configuration.md

mf01, 2020-02-10:
No WiFi AP is available. Should I flash original firmware back? Or is there a another solution to prevent the issue/warning(?) at flashing?

austwhite, 2020-02-10:
Did you use version 0.95 or 1.00?

mf01, 2020-02-10:
In the posted example I used 1.00. But with 0.95 I think it was the same.

klausahrenberg, 2020-04-01:
From v1.00 on a Wifi-AP should be opened after first Power up. If not, maybe the bin file you downloaded is really corrupted and that's why you receive the error message. Please check your downloaded version.

closed 2020-04-03 by klausahrenberg

redelectrical, 2020-05-08:
I'm having the same issues. The device flashes successfully, but with the message "Warning: Image file at 0x0 doesn't look like an image file, so not changing any flash settings.".

No AP appears. Tested using v1.07 and v1.08

If I flash back to original then smartlife AP works so unlikely to be a hardware issue.

Moorey79, 2021-04-04:
I was having this issue.
I tried everything from baud rate to external power.
Nothing helped.
It was doing my head in so I thought I'd make my own (uninformed) opinion of whether it looked like an image or not and opened it in a text editor.
I decided I agreed with esptools pretty quickly.
What it looked like to me was an error message telling me that it couldn't find anything at the web address where I had downloaded the firmware from.
But what would I know? I'd never looked at a tasmota image before.
Eventually I downloaded another one to compare.
The new one looked like rubbish in the text editor.
However esptools seemed to like it alot more.
I guess esptools is a pretty good judge if what art is and if it says something "doesn't look like an image",  give it the benefit if the doubt and trust that the hidden message may have been right in front of you the whole time.