klausahrenberg/WThermostatBeca #71: BHT-6000-USW Support?

diabl0w, 2020-02-04:
Hi!

Can anyone tell me if this firmware supports the BHT-6000-USW? It seems to have multiope names out on the chinese market, but Moes House refers to it as that. It has the terminals like "R" "Y" "W" and "G" that we use in the US. Not sure if other countries use this format. 

It would hook up to several different types of systems. In my case, it would be a conventional centrail air system which all standard thermostats have the following settings: 
A Mode setting: Heat or Cool
A Temperaure setting: integer
A Fan setting: "Always on" or "Auto" 

https://www.moeshouse.com/collections/24v-heat-pump-thermostat/products/moes-smart-wifi-programmable-thermostat-metal-brushed-panel-smart-life-tuya-app-remote-control-works-with-alexa-google-home
https://www.aliexpress.com/i/4000001315029.html

austwhite, 2020-02-04:
I don't think it supports that kind of heat pump configuration but I'll double check with a resource and get back to you.

diabl0w, 2020-02-04:
> I don't think it supports that kind of heat pump configuration but I'll double check with a resource and get back to you.

thank you very much! according to the top review on amazon, https://www.amazon.com/MOES-Programmable-Thermostat-Temperature-Controller/dp/B07V64982S/ref=mp_s_a_1_2?keywords=moes+thermosta,  it looks like it is possible by using Auxiliary port 11 for heat

diabl0w, 2020-02-04:
https://www.yorknow.com/s-sm392505.html

This is the a/c-heating unit I have

diabl0w, 2020-02-04:
[image]

here it says compatible with electric heat which is what I believe I have for the heating part of my central air

austwhite, 2020-02-04:
If your heat/cool switches a single pole switch, such as a 24V line, this may work.  
If you have multiple wires for a compressor it won't.
Example.  If your thermostat uses a single switch to turn the compressor for the air con on and off and the heat is a single switch for a gas or electric furnace, this firmware should work.
Keep a backup of your original firmware though, just in case.
The mode you want to select is BAC-002-ALW.  Just ignore fan speed if your system is single fan speed.

diabl0w, 2020-02-04:
> If your heat/cool switches a single pole switch, such as a 24V line, this may work.
> If you have multiple wires for a compressor it won't.
> Example. If your thermostat uses a single switch to turn the [...]

The heat/cool doesnt really switch one switch like a heat pump orange wire. They just have completely separate terminals... my current thermostat has one red wire to R, one white wire to W1, one yellow wire to Y1, and one green wire to G (fan).

White/W1 controls heat, Yellow/Y1 controls compressor(a/c, cool). The Red/R is the only power wire, there is no separate Rc and Rh (so I believe I will just have to jump these two). So essentially, how I believe this works is extremely simple: Heat Mode = Activate White wire, Cool Mode =  activate Yellow wire. 

This new thermostat has Y1/W1 at the same terminal, so I will have to figure out how to wire it appropriately. I believe i should put Y to Y1/W1, that just leaves a question as to where W should go... I believe the Aux terminal should work but I'm not sure.

diabl0w, 2020-02-04:
I haven't received my device yet, but in theory, I believe that in order to use this thermostat with my system, the firmware needs to be able to trigger emergency mode for heating. Emergency mode would turn Y1/W1 (a/c compressor) "off" and Aux (electric heating) "on"

austwhite, 2020-02-05:
This firmware is designed for the specific thermostats and they do t have aux heat function.
If your system has Aux Heat then a different firmware to this one would be needed.

diabl0w, 2020-02-05:
> This firmware is designed for the specific thermostats and they do not have aux heat function.
> If your system has Aux Heat then a different firmware to this one would be needed.

my system does not have aux heat, it has the configuration I described above (R, Y, W, G, C) 

I am just having a hard time figuring out where the W wire goes since the terminal for W is a "Y/W" terminal and I obviously wouldnt hook up both to the same terminal. I was thinking to use Aux as a workaround to that odd implementation that they used, but maybe there is a better way?

diabl0w, 2020-02-05:
Otherwise, I will just troubleshoot when I get the device.. and see which switches get activated when... and also hook up the device to a logic analyzer and maybe with the help of those smarter than me, get the correct hex codes implemented

austwhite, 2020-02-05:
I think it will become easier when you have the thermostat device to play with.
It sounds like you need assistance more with the wiring of the device rather than firmware at this stage.
That's a bit out of scope for this.
This repo is just for the modified firmware that can uploaded, not for the thermostats in general 🙂.

diabl0w, 2020-02-05:
> I think it will become easier when you have the thermostat device to play with.
> It sounds like you need assistance more with the wiring of the device rather than firmware at this stage.
> That's a bit out of scope for this.
> This repo is just for the modified firmware that can be uploaded, not for the thermostats in general .

I suppose so, I opened another issue that is more relevant to this firmware. There is very little info on this device on the interweb so at least now people can get some page results for their Googling

diabl0w, 2020-02-09:
For anyone googling to here, my solution is as follows to use this device for conventional heating/cooling system:

Fan (Green wire) to Y1/W1,
Heat (white wire) to Aux,
Cool (Yellow wire) to G

IMPORTANT: also go into the settings and turn Fan in Heat mode to "off"

Closing this issue

closed 2020-02-09 by diabl0w