klausahrenberg/WThermostatBeca #73: Different MQTT Topics for state and command

radokristof, 2020-02-09:
Right now, you have to send commands to the same /properties and /schedules topic where the states are represented also.
This is not really how MQTT should work. Right now I can't distinguish which messages are sent by the thermostat and which one is from my commands.
It would be great if commands will be sent for example to /properties/command or something else...

fashberg, 2020-03-11:
Checkout my fork: https://github.com/fashberg/WThermostatBeca
Separated into 'cmnd' and 'stat', like tasmota.

It's working great with homeassistant.
Much easier config

Binary version 1.01b included

radokristof, 2020-03-11:
Thanks! It looks really great, I will update my thermostat to this version soon!

fashberg, 2020-03-11:
If you compile yourself: don’t forget to also use my updated https://github.com/fashberg/WAdapter 

Kind Regards

radokristof, 2020-03-11:
I won't mess with compiling this myself, I will use the built binary. Thanks for letting me know!

austwhite, 2020-03-13:
I am using the binary @fashberg created 1.01b.
So far works well with Home Assistant. 
The MQTT drop/freeze issue is gone by separating topics.

radokristof, 2020-03-14:
PR created

closed 2020-03-14 by radokristof