klausahrenberg/WThermostatBeca #75: Local TZ settings

Hedgehog57, 2020-02-15:
Thanks for Your FW. I really appreciate Your work.

Can I ask You to add a possibility to setup TZ in thermostat? Now the internet access in required.
If after poweroff internet will be inaccesible thermostat will start with wrong time and timezone.
I have installed local NTP server with GPS adapter. So the real time will be available. But not the TZ. And if TZ is bnot available the scheduling in thermosat is useless.

Please, add an option to setup TZ by entering utc_offset directly in the device.

Thanks a lot.

tibo888, 2020-02-17:
i'm agree. My thermostats are not connected to internet so all time zone are not correct.
We appreciate....
Thanks to klausahrenberg, you r king

nimda5, 2020-03-12:
Till this integrated I faked it with nginx. Only static `dst_offset` `raw_offset` `timezone` is used. So can one time open http://worldtimeapi.org/api/ip, copy json, and inset to nginx/apache conf.
My example:
```
server {
        listen       93;
        server_name  localhost;
	location /get_timezone {
			default_type application/json;
			return 200 '{"week_number":11,"utc_offset":"+02:00","utc_datetime":"2020-03-12T12:11:01.078304+00:00","unixtime":1584015061,"timezone":"Europe/Riga","raw_offset":7200,"dst_until":null,"dst_offset":0,"dst_from":null,"dst":false,"day_of_year":72,"day_of_week":4,"datetime":"2020-03-12T14:11:01.078304+02:00","client_ip":"192.168.1.1","abbreviation":"EET"}';
	}
}
```

Hedgehog57, 2020-03-12:
@nimda5 Yes, I did almost the same. I have created static file on my local nginx with a simple content like this:
```
{
"utc_offset":"+03:00",
"timezone":"Europe/Moscow",
"raw_offset":10800,
"dst_until":null,
"dst_offset":0,
"dst_from":null,
"dst":false,
"abbreviation":"MSK"
}
```
That is enough to thermostat to understand my TZ. But that required a local nginx on my raspberry pi 3B+ with lot of load and only 1G RAM. At least we cat set a TZ URL in the firmware...

nimda5, 2020-03-12:
@Hedgehog57 If you ready to patch - seams like not a lot to be done, all in `WClock.h`
before `notifyOnError("Time zone update failed: " + httpCode);` add
```
	lastTimeZoneSync = millis();
	validTime->setBoolean(true);
	notifyOnTimeUpdate();
```
and in `long getRawOffset()` & `long getDstOffset()` return your string values.
I'm not sure it will work, and I don't code.. So it will take time for me, but somedays I will do it...

fashberg, 2020-03-31:
see my fork: https://github.com/fashberg/WThermostatBeca/blob/master/Configuration.md#4-configure-clock-settings
#81

klausahrenberg, 2020-04-01:
Version 1.01 realized the feature now: Time zone sync for offset of NTP-UTC-Time via internet is now optional. Alternatively a fixed offset in minutes can be stored.

closed 2020-04-01 by klausahrenberg

Hedgehog57, 2020-04-01:
Thanks! Works like a charm.