klausahrenberg/WThermostatBeca #77: Adjust temperature

Bl00d-B0b, 2020-02-18:
Hello,

Is it possible to implement temperature adjustments in firmware. as some my thermostats show wrong temperature. 
https://www.youtube.com/watch?v=Eyx_WYHbkGY

it wold be also nice to adjust temperature +/1 0.1 degree, not 0.5, as it's shows in VIDEO.

Ed

klausahrenberg, 2020-04-03:
> Is it possible to implement temperature adjustments in firmware.

It makes no sense because the thermostat has this temperature correction already, here from the manual
[image]
However, it's not possible to modify these settings over Wifi. It's only possible at device.

> it wold be also nice to adjust temperature +/1 0.1 degree, not 0.5

Sorry, this I can not change. This firmware is only the 'Wifi'-Part which is replaced. The original firmware and functions of device can not be modified.

closed 2020-04-03 by klausahrenberg

Bl00d-B0b, 2020-04-05:
Hello,

yes, that's works. +/- 1.0. at least it's OK. maybe you have full table of functions? 

Ed

klausahrenberg, 2020-04-05:
Here ist a link to the manual

https://www.google.com/url?q=http://upload.sunsky-online.com/res/drivers/HC0002_Manual.pdf&sa=U&ved=2ahUKEwiEsKaaq9LoAhUPzqQKHYetCS0QFjAEegQICRAB&usg=AOvVaw3-q2Mv1LQSUg32_F_RmyNd