klausahrenberg/WThermostatBeca #78: Thermostat mqtt reporting and control not working. Only clock messages present.

poedator, 2020-03-03:
Hi, Klaus! This is a very useful project. I am glad I found it and hope to use it in my home. 
Unfortunately I have this issue: 
- After loading firmware (1.00) I was able to enter device web interface and connect it to my network + mqtt server. 
- After boot device sends both clock and thermostat MQTT announcements. 
- But then device only reports clock status message every 5 minutes:
`Mar 03 22:24:09 thermostatbeca_1/things/clock/properties {"ntpServer":"pool.ntp.org","timeZoneServer":"http://worldtimeapi.org/api/ip","epochTime":1583274248,"epochTimeFormatted":"2020-03-03 22:24:08","validTime":true,"timezone":"***************","raw_offset":******,"dst_offset":0}`
It does not send thermostat status info and does not react to any mqtt messages... 
Please help!

austwhite, 2020-03-03:
This is similar the issue I am having with it, except mine will work for a couple of hours then drop to only sending clock data and no temp/function data.

austwhite, 2020-03-10:
Additional note.  I found the MQTT updates stop quicker if the thermostat is operated manually on the device a few times in quick succession

ramon-roca, 2020-03-11:
I think I'm running in the same bug.
I've 8 thermostats, on per room, with a floor heating system, governed by a noderep app on rpi so I'm able to adjust the target temperature through MQTT according to weather forecast and divert solar PV production, so for example increasing target temp while sunny and excess power, so in the evening/night, when there is no PV production, the house is still warm while the heat pump is still off. 
Thats working wonderfuly! No way of doing this without this firm and MQTT support.

I was observing that from time to time some thermostat stopped responding to MQTT. I thought that maybe was having some defect in  a thermostat (and having many) and as a workaround, I just did a hard reboot by simply switching power on/off, I'm also able to do this from the nodered app, but doing for all thermostats (not the one failing). This workaround keeps the system fully functional despite the problem.

Now looking deeper on this, and seeing the comments above, I chanched the power on/off workaround by a selective soft reboot through the web service affecting fustr the thermostat which was not responding, and found that actually all thermostats do have the same behaviour.

That's all can I say from now. Let me know if somehow I can help on testing this....

Bettman66, 2020-03-11:
Since I don't have the problem and did the HardwareRelayMod, it would be nice to know if you are working without the mod.

austwhite, 2020-03-11:
I have done the hardware mod for the state reporting, but the issue was there with and without it.
I have noticed that if you run purely via the MQTT messages and don't send too many at all nice it is more stable, but send a lot of messages at once or start manually controlling with the buttons on the thermostat then it dies, needing a hard boot to get it back.

fashberg, 2020-03-12:
Hey,

if someone wants to test my variant https://github.com/fashberg/WThermostatBeca

@austwhite tested and say his MQTT problem has been gone.

I've separated incoming and outgoing mqtt messages into <topicprefix>/cmnd/things and <topicprefix>/stat/things

Home Assistant ingetration is much more easier:
https://github.com/fashberg/WThermostatBeca#integration-in-home-assisant

Feedback welcome.
Folke

ramon-roca, 2020-03-12:
Hi Foike,
I've just uploaded 1.01b.bin on some of my devices, now leaving home for a couple of days, so over the weekend or so I can be back with the observed differences and confirm if stability improved.
Thanks!
r

austwhite, 2020-03-12:
I can confirm, using latest Home Assistant and latest MosQuiTTo plugin, it has worked for 24 hours without failure.
I have used manual control repeatedly as well as automations.
I have a test thermostat setup using a Beca BHT-002 variant.
Prior to 1.01b if I used manual controls it would stop responding within less than 2 hours and even left alone would stop within 4.
I'll keep monitoring this further, but this is already a big improvement for me.

ramon-roca, 2020-03-18:
Yes, the thermostats are much more stable now.
I had to change my app to the new syntax while sending MQTT commands.

juan11perez, 2020-03-18:
Good day, can someone please offer some guidance on how to configure the clock
I've set as 
http://worldtimeapi.org/api/ip, also as http://worldtimeapi.org/api/timezone/Asia/Dubai but still get UTC time.
Thank you

ramon-roca, 2020-03-20:
Juan, your quetion is not related to this thread. You should open a new link for that.
Anyway what I can say is that http://worldtimeapi.org/api/ip works for me. Look into this urlñ in your browser to see what you are getting, note that is using your intrrnet IP location for getting the timezone, so could be the case that your IP isn't localized.

poedator, 2020-03-20:
Well, unlike some of you here, I see no improvements. I tried 1.01b and 1.03b firmwares from @fashberg - same situation: only updates of time and config posted to mqtt at start and no regular temperature updates. Besides, the thermostat  is not even reacting on incoming mqtt messages supposed to set the temperature... Is there anything else I could try?

fashberg, 2020-03-20:
@poedator hi,

With my 1.03b the output topic should have changed to  thermostatbeca_1/stat/things/xxx/properties

Commands can be send to 
thermostatbeca_1/cmnd/things/clock/properties or thermostatbeca_1/cmnd/things/thermostat/properties

With 1.03b please try
terminal 1 - monitor: 
# mosquitto_sub - h mqttserver -t thermostatbeca_1/# -v
Terminal 2 - command
# mosquitto_pub - h mqttserver -t thermostatbeca_1/things/thermostat/properties/logMcu -m true

Please show output of mqtt monitor terminal.
Debuglog ends with reboot of thermostat or set logMcu to false

Kind regards
Folke

austwhite, 2020-03-21:
The separated command and state topics stopped the freezing.
1.02b made real time temp updates happen if you use the correct state topics.
1.03b that from xed scheduling did break something as it crashed after about 12hours so I went back to 1.02b as I don't use scheduling via MQTT.  I use HA for all that stuff.

ramon-roca, 2020-03-21:
I just tested 1.01b for now.
In my case, I was using MQTT for getting the readings every 20s and adjust to PV readings. so not waiting for the 5m message. 
I got confused also at the beginning for not getting the messages, but after some research found that was because of the change on the topic syntax.

fashberg, 2020-03-21:
@austwhite 
Hmm, interesting.
My 5 thermostates still working fine.
Yes, please share all you have. Software and Version of mqtt server, hass config including thermostate relevant automation and screenshots of thermostate config (network, thermostat).

What happens if you publish messages in that buggy state to thermostate? Any responses?
Is it reachable using http?
Maybe also try to enable logMcu as described here https://github.com/klausahrenberg/WThermostatBeca/issues/78#issuecomment-601925386 

Kind regards

austwhite, 2020-03-21:
@fashberg I'll get a look at all that during the week.  Just got a few things on during the weekend 🙂

juan11perez, 2020-03-21:
@ramon-roca thank you. didnt want to open a new thread as it's not a bug. Im just unsure Im doing the clock setting correctly. I set it to http://worldtimeapi.org/api/ip and it finds my location correctly, however the clock stays in UTC.

@fashberg, thank you for creating/improving this component. Im running it on a ALW and it's working very well on 1.02b

poedator, 2020-03-22:
@fashberg. Thank you for the test suggestion. Here is my MQTT output:

at boot: 
`Mar 23 00:21:14 devices/clock {"url":"http://192.168.1.219/things/clock","ip":"192.168.1.219","topic":"thermostatbeca_1/things/clock"}
Mar 23 00:21:14 devices/thermostat {"url":"http://192.168.1.219/things/thermostat","ip":"192.168.1.219","topic":"thermostatbeca_1/things/thermostat"}
Mar 23 00:21:14 thermostatbeca_1/stat/things/clock/properties {"ntpServer":"pool.ntp.org","timeZoneServer":"http://worldtimeapi.org/api/ip","epochTime":1584922874,"epochTimeFormatted":"2020-03-23 00:21:14","validTime":true,"timezone":"Europe/Moscow","raw_offset":10800,"dst_offset":0}`

then after entering `logMcu true` command:
 `Mar 23 00:23:25 thermostatbeca_1/cmnd/things/thermostat/properties/logMcu true
Mar 23 00:23:30 thermostatbeca_1/log {"log":"sending heartBeatCommand"}
Mar 23 00:23:30 thermostatbeca_1/log {"log":"commandCharsToSerial: 55 aa 00 00 00 00"}`
the last 2 lines repeat every 10 seconds.

Here is output from `http://[IP_ADDRESS]/things/thermostat`:
`{"name":"thermostat","href":"/things/thermostat","@context":"https://iot.mozilla.org/schemas","@type":["Thermostat"],"properties":{"temperature":{"title":"Actual","type":"number","readOnly":true,"unit":"celsius","@type":"TemperatureProperty","href":"/things/thermostat/properties/temperature"},"targetTemperature":{"title":"Target","type":"number","unit":"celsius","multipleOf":0.50,"@type":"TargetTemperatureProperty","href":"/things/thermostat/properties/targetTemperature"},"deviceOn":{"title":"Power","type":"boolean","@type":"OnOffProperty","href":"/things/thermostat/properties/deviceOn"},"schedulesMode":{"title":"Schedules","type":"string","enum":["off","auto"],"@type":"ThermostatSchedulesModeProperty","href":"/things/thermostat/properties/schedulesMode"},"mode":{"title":"Mode","type":"string","enum":["off","auto","heat"],"@type":"ThermostatModeProperty","href":"/things/thermostat/properties/mode"}}}`
The thermostat itself shows current temperature of 21c and preset at 25c.

klausahrenberg, 2020-04-03:
Version 1.02 should fix the issue. Tested over 12 hours. If still present, please reopen and tell me your exact model please.

closed 2020-04-03 by klausahrenberg