klausahrenberg/WThermostatBeca #80: Set systemMode not working on BAC-6000-ELW

zalke, 2020-03-08:
I have the above referenced model, all setting work except changing from heat to cool,

I tried with Tasmota and the following commands work just fine

```TuyaSend4 102,00```
Sets the mode to Cool

```TuyaSend4 102,01```
Sets the systemMode to Heat

klausahrenberg, 2020-04-03:
Please try version 1.02. If your device reported the correct states, but you was not able to set the modes, than the new version will help.

zalke, 2020-04-06:
@klausahrenberg 

I'm please to inform that the current version it is working correctly.

Thanks so much for your great work,

BTW is the any way to change from Celsius to Fahrenheit on these thermostats?

klausahrenberg, 2020-04-06:
That's nice to hear that it's working now. 

According to the manual, you can't switch device to  Fahrenheit

closed 2020-04-06 by klausahrenberg