klausahrenberg/WThermostatBeca #82: Help in scheduler response

nimda5, 2020-03-12:
scheduler for mqtt and mozilla have json structure like below.
I don't understand what means w1h ; w1t ; u1h ; u1t; a1h; a1t
Maybe for json this is regular think, but not for me at least ;) 

```
{
w1h | "06:00"
w1t | 20
w2h | "08:00"
w2t | 15
w3h | "11:30"
w3t | 15
w4h | "13:30"
w4t | 15
w5h | "17:00"
w5t | 22
w6h | "22:00"
w6t | 15
a1h | "06:00"
a1t | 20
a2h | "08:00"
a2t | 20
a3h | "11:30"
a3t | 20
a4h | "13:30"
a4t | 20
a5h | "17:00"
a5t | 20
a6h | "22:00"
a6t | 15
u1h | "06:00"
u1t | 20
u2h | "08:00"
u2t | 20
u3h | "11:30"
u3t | 20
u4h | "13:30"
u4t | 20
u5h | "17:00"
u5t | 20
u6h | "22:00"
u6t | 15
 }
```

fashberg, 2020-03-12:
w = Workday
a = sAturday
u = sUnday

h = hour (+minute)
t = temperature

austwhite, 2020-03-12:
Makes sense now.
Probably the best way to do it given the thermostat separates Saturday and Sunday

nimda5, 2020-03-12:
Correct me I wrong on original example for working day:
06:00 - 08:00: 20 deg
08:00 - 11:30: 15 deg
--//--
17:00 - 22:00: 22 deg

I was lost understanding what is number in the middle of, now I think got:
<Workday/sAturday/sUnday/><ID><hour/temperature>

@fashberg Thanks for your port, will try it. Just small improvements to make: add for the web config BHT-002-GCLW, now it's only in readme.

austwhite, 2020-03-12:
If an option is read only, you may need to fully reset your device.
The options are still selectable for me

Hedgehog57, 2020-03-12:
@nimda5 the number in the middle is the interval number. You have 6 intervals per day. For example
```
- night
- morning
- leaving at work
- day
- returning at home
- evening
```
Or You can cut the 24 hours like You want.

closed 2020-03-12 by nimda5