klausahrenberg/WThermostatBeca #83: BHT-002GBLW a slight buzzing sound when heating is off.

Lippo89, 2020-03-13:
Hi,

At first thanks for the firmware, it's working nicely.

I'm just bit worried about this slight buzzing sound is it normal or is my wiring wrong some how? My house have ceiling heater elements and old thermostat was bi-metal type. There comes three wires for the thermostat and colors are blue,black and brown. I tested there was voltage only in black wire, so then I fixed black to terminal 4 and blue into 3. The brown one I put into load side terminal 1 and terminal 2 is empty. The external temperature sensor is not installed and I changed in settings to use only internal sensor.

Otherwise this setup works heater elements are heating when temperature drops and goes of when gets the desired temperature. The buzzing sound stops when heating is on.

BR,
Lippo

austwhite, 2020-03-13:
I get no buzzing sounds.
There is nothing in the firmware that would cause this.
Sounds like a wiring issue or maybe a faulty device.

Lippo89, 2020-03-14:
Ok, thanks! Then it might be wiring issue or the faulty device. But what might be wrong with wiring and can cause this?

austwhite, 2020-03-14:
Unsure.  I'd just be guessing right now as I am not there.
Probably the first thing to check would be to make sure all the wires are tight and nothing is loose.
It's a little out of scope for us here 🙂

Lippo89, 2020-03-14:
Yep, I just thought maybe it's common that this device makes a slight sound. It's not loud sound and you can hear it only when you put ear close to the thermostat. I tested now with only neutral and live wires and it still makes the noise. So I think it's not wiring issue and I have just faulty device. 

Thanks for the help and sorry for this off-topic.

closed 2020-03-14 by Lippo89

pedrosimao, 2020-10-05:
@Lippo89 I have the exact same issue, but with the original firmware (still didin't try to install this firmware).
The thermostat is installed close to my desk, (less than 1m from my right ear). So I hear the noise while working and its super annoying.
When I am 2m far from it I can't hear it anymore, but well, changing my desk position is not an option.
Did you find a solution to the problem? Changing the device did work?

Lippo89, 2020-10-09:
@pedrosimao I didn't change the device and it has been working just fine otherwise. It makes still that slight sound, but it's not big problem for me. I also ordered couple more of these, but those aren't installed yet.