klausahrenberg/WThermostatBeca #84: Floureon HY08WE and WThermostatBeca

javier6d2, 2020-03-14:
Hi  klausahrenberg!

I have installed your firmware in a thermostat Floureon HY08WE, I get sucess, however, the value of the sensor temp. showed in the State report are not correct, it is 5 times greater:

Thermo.___json messg.
T(ºC)______ values
16---------------80
17---------------85
18---------------90
19---------------95
20--------------100
21--------------105
21.5-----------107.5
22--------------110

Properties:
{"idx":"560","ip":"192.168.130.211","firmware":"1.01beta","temperature":112.50,"targetTemperature":110.00,"deviceOn":true,"schedulesMode":"auto","mode":"auto","state":"off"}

My MQTT-server received some unknown messages:

14/3/2020 14:40:01node: ba383984.9ef16home/room/mcucommand : msg.payload : string[46]
"{"unknown":"55 aa 01 07 00 05 0c 05 00 01 00"}"
14/3/2020 14:40:01node: ba383984.9ef16home/room/mcucommand : msg.payload : string[55]
"{"unknown":"55 aa 01 07 00 08 67 02 00 04 00 00 00 00"}"
14/3/2020 14:40:01node: ba383984.9ef16home/room/mcucommand : msg.payload : string[55]
"{"unknown":"55 aa 01 07 00 08 68 02 00 04 00 00 00 03"}"
14/3/2020 14:40:01node: ba383984.9ef16home/room/mcucommand : msg.payload : string[55]
"{"unknown":"55 aa 01 07 00 08 69 02 00 04 00 00 00 14"}"
14/3/2020 14:40:01node: ba383984.9ef16home/room/mcucommand : msg.payload : string[46]
"{"unknown":"55 aa 01 07 00 05 6a 01 00 01 01"}"
14/3/2020 14:40:01node: ba383984.9ef16home/room/mcucommand : msg.payload : string[46]
"{"unknown":"55 aa 01 07 00 05 6b 01 00 01 01"}"
14/3/2020 14:40:01node: ba383984.9ef16home/room/mcucommand : msg.payload : string[46]
"{"unknown":"55 aa 01 07 00 05 6c 01 00 01 01"}"
14/3/2020 14:40:01node: ba383984.9ef16home/room/mcucommand : msg.payload : string[55]
"{"unknown":"55 aa 01 07 00 08 6d 02 00 04 ff ff ff f6"}"
14/3/2020 14:40:02node: ba383984.9ef16home/room/mcucommand : msg.payload : string[55]
"{"unknown":"55 aa 01 07 00 08 6e 02 00 04 00 00 00 0a"}"
14/3/2020 14:40:02node: ba383984.9ef16home/room/mcucommand : msg.payload : string[55]
"{"unknown":"55 aa 01 07 00 08 6f 02 00 04 00 00 00 02"}"
14/3/2020 14:40:02node: ba383984.9ef16home/room/mcucommand : msg.payload : string[55]
"{"unknown":"55 aa 01 07 00 08 70 02 00 04 00 00 00 2d"}"
14/3/2020 14:40:02node: ba383984.9ef16home/room/mcucommand : msg.payload : string[55]
"{"unknown":"55 aa 01 07 00 08 71 02 00 04 00 00 00 05"}"
14/3/2020 14:40:02node: ba383984.9ef16home/room/mcucommand : msg.payload : string[55]
"{"unknown":"55 aa 01 07 00 08 72 02 00 04 00 00 00 23"}"
14/3/2020 14:40:02node: ba383984.9ef16home/room/mcucommand : msg.payload : string[55]
"{"unknown":"55 aa 01 07 00 08 73 02 00 04 00 00 00 05"}"
14/3/2020 14:40:02node: ba383984.9ef16home/room/mcucommand : msg.payload : string[46]
"{"unknown":"55 aa 01 07 00 05 74 04 00 01 00"}"
14/3/2020 14:40:02node: ba383984.9ef16home/room/mcucommand : msg.payload : string[46]
"{"unknown":"55 aa 01 07 00 05 75 04 00 01 00"}"
14/3/2020 14:40:02node: ba383984.9ef16
home/room/mcucommand : msg.payload : string[46]
"{"unknown":"55 aa 01 07 00 05 76 04 00 01 00"}"

Would be very nice get the correct MQTT messages!

Kind regards

klausahrenberg, 2020-04-03:
Hi, I checked the issue. It seems that your model is working with another factor to multiply the raw value to get the resulting. The original factor at my device is 2, so at your device the factor seems to be 10. However, with factor 10 I will have a buffer overrun above 26 degree (10*26 > 255) - and 255 is a magical border for byte values. To correct this, I have to do bigger changes in code. And the schedules data have only 1 byte on my device - this must be different at your device. This maybe explains your unknown command messages above.
To summarize, I will not implement a support for your device. Sorry for no better news.

klausahrenberg, 2020-04-22:
Hi, I implemented a dividor by 10 in v1.05. Please select your model in device configuration and test. Your unknown mcu commands are not implemented yet. This will follow, if temperature works for you. Please check

javier6d2, 2020-04-22:
You did it, now I can see the correct temperatures. Through MQTT I can turn the device on or off and change the setpoint temperature. The schedule mode seems to work the other way around, if I send the command  ` { "schedulesMode":"off" }`  shedule mode is activated.

I still get messages with failures but the main things seems to work fine.

zu3st-de, 2020-07-25:
hi, is it possible to make a hardware mod and get state working?

[image]

the most things seem to work, but I get many  {"error":"Unknown MCU command"}

```
home/gz/clock/properties {"ntpServer":"pool.ntp.org","timeZoneServer":"http://worldtimeapi.org/api/ip","epochTime":1595683165,"epochTimeFormatted":"2020-07-25 13:19:25","validTime":true,"timezone":"Europe/Berlin"}
home/gz/thermostat/properties {"idx":"gz_thermostat","ip":"192.168.178.88","firmware":"1.11","temperature":25.00,"targetTemperature":21.00,"deviceOn":false,"schedulesMode":"auto","locked":true,"floorTemperature":0.00}
home/gz/thermostat/properties/ {"idx":"gz_thermostat","ip":"192.168.178.88","firmware":"1.11","temperature":25.00,"targetTemperature":21.00,"deviceOn":false,"schedulesMode":"auto","locked":true,"floorTemperature":0.00}
home/gz/thermostat/properties//temperature 24.00
home/gz/thermostat/properties//targetTemperature 15.00
home/gz/thermostat/properties//deviceOn false
home/gz/thermostat/properties//schedulesMode "off"
home/gz/thermostat/properties//locked true
home/gz/thermostat/properties//floorTemperature 0.00
home/gz/thermostat/properties/temperature 24.00
home/gz/thermostat/properties/targetTemperature 15.00
home/gz/thermostat/properties/deviceOn false
home/gz/thermostat/properties/schedulesMode "off"
home/gz/thermostat/properties/locked true
home/gz/thermostat/properties/floorTemperature 0.00
home/gz {"error":"Unknown MCU command"}
home/gz {"error":"Unknown MCU command"}
home/gz {"error":"Unknown MCU command"}
home/gz {"error":"Unknown MCU command"}
home/gz {"error":"Unknown MCU command"}
home/gz {"error":"Unknown MCU command"}
home/gz {"error":"Unknown MCU command"}
home/gz {"error":"Unknown MCU command"}
home/gz {"error":"Unknown MCU command"}
home/gz {"error":"Unknown MCU command"}
home/gz {"error":"Unknown MCU command"}
home/gz {"error":"Unknown MCU command"}
home/gz {"error":"Unknown MCU command"}
home/gz {"error":"Unknown MCU command"}
home/gz {"error":"Unknown MCU command"}
home/gz {"error":"Unknown MCU command"}
home/gz {"error":"Unknown MCU command"}
home/gz {"error":"Unknown MCU command"}
```

klausahrenberg, 2021-01-13:
Hi, sorry for the delay. Please test v1.20a and report the unknown MCU commands to me. I need the concrete commands that are reported to the MQTT broker, starting with "55 aa ..."

klausahrenberg, 2021-01-14:
Check binary WThermostat_1.20b_alpha, I added support for schedules mode (DpId 4). Should be Manual, Auto, Holiday, Hold (acc. to manual: hold temp for 1 h, and then switch back to auto?). Please check

javier6d2, 2021-01-16:
Hello, I have update the firmware but if I select the thermostat model like Floureon the thermostat stop working the Wi-Fi. I have to erase flash memory and upload the firmware again, but can not choose floureon here:
[image]

klausahrenberg, 2021-01-17:
Sorry, you are right - there was a mistake in code. When you switched to Flureon the ESP crasehd. I fixed it now in 1.20c. Please try again

Trashmee, 2021-02-07:
Hi,

just flashed my first bht-002 with your firmware (thanks a lot!) - get those messages, too:

```
bht-001-01/clock/properties {"ntpServer":"pool.ntp.org","timeZoneServer":"http://worldtimeapi.org/api/ip","epochTime":1612723987,"epochTimeFormatted":"2021-02-07 18:53:07","validTime":true,"timezone":"Europe/Berlin"}
bht-001-01/thermostat/properties {"idx":"bht-001-01","ip":"192.168.180.180","alive":true,"firmware":"1.20l","temperature":21.50,"targetTemperature":20.00,"floorTemperature":0.00,"deviceOn":true,"schedulesMode":"auto","locked":false,"state":"heating"}
bht-001-01 {"error":"Unknown MCU command: 55 aa 01 07 00 05 68 01 00 01 01"}
[..]
bht-001-01 {"error":"Unknown MCU command: 55 aa 01 07 00 05 68 01 00 01 01"}
```

Regards,
Alex

NiOsY, 2021-03-19:
I also get the same 1 per second. (fw 1.20m) 

<my_mqtt_topic>
{"error":"Unknown MCU command: 55 aa 01 07 00 05 68 01 00 01 01"}

other: Can the state report be modified to be submitted every 1 minute witout recompille binary? 

Regards,
Norbert

titiviking, 2022-03-04:
Experiencing the same on 1.20n. 
(logging on web UI not available, unable to post log)

From MQTT:
`Message 125 received on home/Livingroom/Thermostat/ at 9:50 AM:
{
    "error": "Unknown MCU command: 55 aa 01 07 00"
}
QoS: 0 - Retain: false
Message 124 received on home/Livingroom/Thermostat/ at 9:50 AM:
{
    "error": "Unknown MCU command: 55 aa 01 07 00"
}
QoS: 0 - Retain: false
Message 123 received on home/Livingroom/Thermostat/ at 9:50 AM:
{
    "error": "Unknown MCU command: 55 aa 01 07 00"
}
QoS: 0 - Retain: false
Message 122 received on home/Livingroom/Thermostat/ at 9:50 AM:
{
    "error": "Unknown MCU command: 55 aa 01 07 00"
}
QoS: 0 - Retain: false
Message 121 received on home/Livingroom/Thermostat/ at 9:50 AM:
{
    "error": "Unknown MCU command: 55 aa 01 07 00"
}
QoS: 0 - Retain: false
Message 120 received on home/Livingroom/Thermostat/ at 9:50 AM:
{
    "error": "Unknown MCU command: 55 aa 01 07 00"
}
QoS: 0 - Retain: false
Message 119 received on home/Livingroom/Thermostat/ at 9:50 AM:
{
    "error": "Unknown MCU command: 55 aa 01 07 00"
}
QoS: 0 - Retain: false
Message 118 received on home/Livingroom/Thermostat/ at 9:49 AM:
{
    "error": "Unknown MCU command: 55 aa 01 07 00"
}
QoS: 0 - Retain: false
Message 117 received on home/Livingroom/Thermostat/ at 9:49 AM:
{
    "error": "Unknown MCU command: 55 aa 01 07 00"
}
QoS: 0 - Retain: false
Message 116 received on home/Livingroom/Thermostat/ at 9:49 AM:
{
    "error": "Unknown MCU command: 55 aa 01 07 00"
}
QoS: 0 - Retain: false`

erew123, 2022-12-19:
Hi. Thought Id join in on this thread (if its still live). Ive tried 1.25beta with a HY08WE-04 (Underfloor heating model with an external temperature sensor connected).

Here's the first 20-30 messages Im getting from MQTT

thermostat_6780867/thermostat/properties
{"idx":"thermostat_6780867","ip":"192.168.1.13","alive":false}
thermostat_6780867/thermostat/properties
{"idx":"thermostat_6780867","ip":"192.168.1.13","alive":true}
thermostat_6780867/clock/properties
{"ntpServer":"pool.ntp.org","timeZoneServer":"http://worldtimeapi.org/api/ip","epochTime":1671473865,"epochTimeFormatted":"2022-12-19 18:17:45","validTime":true,"timezone":"Europe/London"}
thermostat_6780867
{"error":"Unknown MCU command: 55 aa 01 07 00"}
thermostat_6780867
{"error":"Unknown MCU command: 55 aa 01 07 00"}
thermostat_6780867
{"error":"Unknown MCU command: 55 aa 01 07 00 08 67 02"}
thermostat_6780867
{"error":"Unknown MCU command: 55 aa 01 07 00 08 68 02"}
thermostat_6780867
{"error":"Unknown MCU command: 55 aa 01 07 00 08 69 02"}
thermostat_6780867
{"error":"Unknown MCU command: 55 aa 01 07 00"}
thermostat_6780867
{"error":"Unknown MCU command: 55 aa 01 07 00"}
thermostat_6780867
{"error":"Unknown MCU command: 55 aa 01 07 00"}
thermostat_6780867
{"error":"Unknown MCU command: 55 aa 01 07 00 08 6d 02"}
thermostat_6780867
{"error":"Unknown MCU command: 55 aa 01 07 00 08 6e 02"}
thermostat_6780867
{"error":"Unknown MCU command: 55 aa 01 07 00 08 6f 02"}
thermostat_6780867
{"error":"Unknown MCU command: 55 aa 01 07 00 08 70 02"}
thermostat_6780867
{"error":"Unknown MCU command: 55 aa 01 07 00 08 71 02"}
thermostat_6780867
{"error":"Unknown MCU command: 55 aa 01 07 00 08 72 02"}
thermostat_6780867
{"error":"Unknown MCU command: 55 aa 01 07 00 08 73 02"}
thermostat_6780867
{"error":"Unknown MCU command: 55 aa 01 07 00"}
thermostat_6780867
{"error":"Unknown MCU command: 55 aa 01 07 00"}
thermostat_6780867
{"error":"Unknown MCU command: 55 aa 01 07 00"}
thermostat_6780867
{"error":"Unknown MCU command: 55 aa 01 07 00"}
thermostat_6780867
{"error":"Unknown MCU command: 55 aa 01 07 00"}
thermostat_6780867
{"error":"Unknown MCU command: 55 aa 01 07 00"}
thermostat_6780867
{"error":"Unknown MCU command: 55 aa 01 07 00 08 67 02"}
thermostat_6780867
{"error":"Unknown MCU command: 55 aa 01 07 00 08 68 02"}
thermostat_6780867
{"error":"Unknown MCU command: 55 aa 01 07 00 08 69 02"}
thermostat_6780867
{"error":"Unknown MCU command: 55 aa 01 07 00"}
thermostat_6780867
{"error":"Unknown MCU command: 55 aa 01 07 00 08 73 02"}
thermostat_6780867