klausahrenberg/WThermostatBeca #85: Feature request. Create topics in mqtt with `retain` flag

bessarabov, 2020-03-17:
First of all, thank you for this project!
I'm using it with Home Assistant and I'm very happy with it.

But there is one small thing that I want to be changed. Here is what I get in mqtt topic:

```
{
  "deviceOn": true,
  "ecoMode": false,
  "firmware": "1.00",
  "floorTemperature": 15.5,
  "idx": "thermostatbeca_13610097",
  "ip": "192.168.31.40",
  "locked": false,
  "schedulesMode": "off",
  "targetTemperature": 3,
  "temperature": 8
}
```

Everything is super here, expect that this topic does not have `retain` flag. It means that when I restart Home Assistant for some time Home Assistant does not know about the state of the thermostat. And because of that there are some gaps on the graph with temperature:

[image]

I'm sure it is possible to fix on the side of my Home Assistant configs (maybe to use the last value if there is no current value), but I think that the more propper way to solve this issue is to add `retain` flag to the topics that this soft is sending to mqtt server. I don't know any drawbacks of such solution, but if there are any, maybe this feature can be implemented with some config settings in the web interface. By defaul everything is working as it is working now, but with some checkbox set the flag `retained` is being set (but, from my point of view I think that this flag should be added without any settings, I can't image when it can do some harm)

austwhite, 2020-03-17:
If you turn the thermostat off and on after a restart of home Assistant it will send the packet immediately and update.
The issue is there is a 5 minute delay between sending updates unless the thermostat switches modes.
Personally, I don't care about the 5 minute gap in the graphs.  Graphs are not critical to me, but I get your point.
You can't really set retain in the HA config though.  That really needs to come from the MQTT in the firmware.
There is a version that updates every 10 seconds listed in one of the issues, but that hammers the NTP server a little bit you could try that.
There is a request to investigate speeding up updates in another issue, along with a request that separates state and command topics, as this can cause issues

bessarabov, 2020-03-17:
> If you turn the thermostat off and on after a restart of home Assistant it will send the packet immediately and update.

The other way to update topic is to send message with empty payload to <your_topic>/things/thermostat/properties

austwhite, 2020-03-17:
Indo think having the option to set retail bed would be good though, but do make it as an option and not always on.

bessarabov, 2020-03-17:
Adding such and option perfectly suits my needs. But I'm just curious, why somebody would prefer not to have this topic with `retain` flag? How can this flag affect something is a bad way?

austwhite, 2020-03-17:
There are some implementations that don't need retain. 
This firmware is not dedicated to HA.

bessarabov, 2020-03-17:
> There are some implementations that don't need retain.

Are there any examples? For the moment I can't image any real world situation where retain flag will break something.

fashberg, 2020-03-17:
Hi
using my fork  you'll get this feature.
Check out https://github.com/fashberg/WThermostatBeca  (also you need my version of  WAadpter, or just try binary 1.02b)

No response from klausahrenberg on my pull request.

If you start e.g.
# mosquitto_sub -h mqtt -v  -t 'home/room/#'
you will get instantly:
home/room/stat/things/thermostat/properties {"idx":"thermostat_room","ip":"10.10.200.43","firmware":"1.02b","logMcu":false,"temperature":24.50,"targetTemperature":18.00,"deviceOn":true,"schedulesMode":"auto","ecoMode":false,"locked":false,"floorTemperature":0.00,"mode":"auto"}

You need a separate MQTT broker. The HASS internal one drops retained messages on reboot.
Also i've changed the interval vom 300 to 60 seconds

austwhite, 2020-03-17:
Just a note on above comment.  If you use the MosQuiTTo HA add-on your retains are respected.  It's only the inbuilt broker that has issues.
Great work @fashberg .
I'll run with 1.02 and see how it goes

austwhite, 2020-03-17:
@fashberg Looks good so far.  Flashed quickly OTA and the firmware (1.02b) seems stable.
I'll use (and abuse) it for a couple of days and report back on the pull request #81 .
So far, the changes you made are great improvements to this firmware, especially for HA users.

bessarabov, 2020-03-18:
> using my fork you'll get this feature. Check out https://github.com/fashberg/WThermostatBeca (also you need my version of WAadpter, or just try binary 1.02b)

Thank you! I appreciate all the work that you've done. But there are so many changes that I'm afraid to spend time trying to use that code. For the moment `WThermostat_1.00.bin` suits me pretty well. I've configured everything in Home Assistant and it is working (my HA solution is far from perfect, but I can manage everything that I need from HA) I'm not sure, but it looks like there are breaking changes in your code and I will not be able to just replace `WThermostat_1.00.bin` with compiled version of your code, but I need to change how my HA interacts with mqtt server.

austwhite, 2020-03-18:
You don't have to use this code if the other one works.  This is for people that are having issues with the other one

austwhite, 2020-03-26:
@bessarabov 
Did you update to use 1.02b (or now 1.03b)?
If so, did you get it working in HA?

Just in case you didn't, or if someone else wants to look at this later, the yaml in HA that I use (using the hardware mod to show heating and idle) is like this: 

```
climate:
  - platform: mqtt
    name: Central heating
    temperature_command_topic: "home/centralheat/cmnd/things/thermostat/properties/targetTemperature"
    temperature_state_topic: "home/centralheat/stat/things/thermostat/properties"
    temperature_state_template: "{{ value_json['targetTemperature'] }}"
    current_temperature_topic: "home/centralheat/stat/things/thermostat/properties"
    current_temperature_template: "{{ value_json['temperature'] }}"
    away_mode_command_topic: "home/centralheat/cmnd/things/thermostat/properties/ecoMode"
    away_mode_state_topic: "home/centralheat/stat/things/thermostat/properties"
    away_mode_state_template: "{{ value_json['ecoMode'] }}"
    payload_on: "true"
    payload_off: "false"
    mode_command_topic: "home/centralheat/cmnd/things/thermostat/properties/mode"
    action_topic: "home/centralheat/stat/things/thermostat/properties"
    action_template: >
      {% if value_json.state == 'heating' %}
      heating
      {% elif value_json.mode == 'off' %}
      off
      {% else %}
      idle
      {% endif %}
    mode_state_topic: "home/centralheat/stat/things/thermostat/properties"
    mode_state_template: "{{ value_json['mode'] }}"
    modes:
      - "heat"
      - "auto"
      - "off"
    min_temp: 5
    max_temp: 35
    temp_step: 0.5
    precision: 0.5
```

klausahrenberg, 2020-04-03:
> Create topics in mqtt with `retain` flag

Hi, it's realized in version 1.02

closed 2020-04-03 by klausahrenberg

bessarabov, 2020-04-03:
@klausahrenberg Thank you!