klausahrenberg/WThermostatBeca #86: Not possible to set fan and system mode on bac-002

martonmodis13, 2020-03-18:
Hi!
First of all thanks for your great job,firmware on my bht-002 devices work perfectly.
The problem:
From Home assistant with mqtt I can't set the systemMode and fanMode.(on bac-002)
Model bht-002 work correctly,but iI think bac-002 cannot be integrated into Home assistant yet.
Any ideas or solutions?

JoostHman, 2020-03-30:
Just want to chime in here on the issue with the bac-002. The communication via MQTT does seem to work, however systemMode and fanMode do not actually change anything on the termostat itself.

I also wanted to mention that I did the original firmware conversion BAC-002-ALW via Tuya Convert and it worked without problems!

austwhite, 2020-03-30:
@JoostHman 
Did you update your topics for all actions including fan_mode topics?
I can't test with mine as my situation is a gas boiler and hydronic with no fan coils.

JoostHman, 2020-03-31:
Hi Austwhite, thanks for the quick reaction. So my termostat has been on heating (sun icon) all the time while trying to change the system mode setting:

**These are the MQTT messages.**
 
_Message 66 received on homeassistant/hvac/tu/things/thermostat/properties at 9:02 AM:
{
    "idx": "thermostat_upstairs",
    "ip": "192.168.1.177",
    "firmware": "1.00",
    "temperature": 20,
    "targetTemperature": 5,
    "deviceOn": true,
    "schedulesMode": "off",
    "ecoMode": false,
    "locked": false,
    "systemMode": "cool",
    "fanMode": "low"
}
QoS: 0 - Retain: false
Message 65 received on homeassistant/hvac/tu/things/thermostat/properties/systemMode at 9:02 AM:
cool
QoS: 0 - Retain: false
Message 64 received on homeassistant/hvac/tu/things/thermostat/properties/deviceOn at 9:02 AM:
true
QoS: 0 - Retain: false
Message 63 received on homeassistant/hvac/tu/things/clock/properties at 9:01 AM:
{
    "ntpServer": "pool.ntp.org",
    "timeZoneServer": "http://worldtimeapi.org/api/ip",
    "epochTime": 1585645286,
    "epochTimeFormatted": "2020-03-31 09:01:26",
    "validTime": true,
    "timezone": "Europe/Madrid",
    "raw_offset": 3600,
    "dst_offset": 3600
}
QoS: 0 - Retain: false
Message 62 received on homeassistant/hvac/tu/things/thermostat/properties at 9:00 AM:
{
    "idx": "thermostat_upstairs",
    "ip": "192.168.1.177",
    "firmware": "1.00",
    "temperature": 20,
    "targetTemperature": 5,
    "deviceOn": true,
    "schedulesMode": "off",
    "ecoMode": false,
    "locked": false,
    "systemMode": "fan_only",
    "fanMode": "low"
}
QoS: 0 - Retain: false
Message 61 received on homeassistant/hvac/tu/things/thermostat/properties/systemMode at 9:00 AM:
fan_only
QoS: 0 - Retain: false_

**Configuration.yaml entry:**

climate:
 - platform: mqtt
   name: thermostat_upstairs
   power_command_topic: "homeassistant/hvac/tu/things/thermostat/properties/deviceOn"
   payload_on: "true"
   payload_off: "false"
   temperature_command_topic: "homeassistant/hvac/tu/things/thermostat/properties/targetTemperature"
   temperature_state_topic: "homeassistant/hvac/tu/thermostat/properties"
   temperature_state_template: "{{ value_json['targetTemperature'] | round(1) }}"
   current_temperature_topic: "homeassistant/hvac/tu/things/thermostat/properties"
   current_temperature_template: "{{states('sensor.thermostat_current_temperature')}}"
   mode_command_topic: "homeassistant/hvac/tu/things/thermostat/properties/systemMode"
   mode_state_topic: "homeassistant/hvac/tu/things/thermostat/properties"
   mode_state_template: >
            {% if value_json.deviceOn != true %}
            off
            {% elif value_json.schedulesMode == 'auto' %}
            auto
            {% else %}
            "{{value_json.systemMode}}"
            {% endif %}
   modes: 
          - "heat"
          - "cool" 
          - "fan_only"
          - "off"
   fan_modes:
         - "auto"
         - "low"
         - "medium"
         - "high"
   fan_mode_state_topic: "homeassistant/hvac/tu/things/thermostat/properties"
   fan_mode_command_topic: "homeassistant/hvac/tu/things/thermostat/properties/fanMode"
   fan_mode_state_template: "{{value_json.fanMode}}"
   min_temp: 5
   max_temp: 25
   temp_step: 0.5
   precision: 0.5
   retain: true

fashberg, 2020-03-31:
Hi!
I saw that the systemMode and fanMode have not been updated from Wifi Module to MCU.
I fixed it in my fork, releas 1.05b https://github.com/fashberg/WThermostatBeca 

Please try and give feedback!

Kind Regards
Folke

JoostHman, 2020-04-01:
Fantastic, this works now! One question regarding the state flag in the
message below. Why would this say "cooling" if systemMode is "heat" what is
the purpose?

Message 34 received on
homeassistant/hvac/tu/stat/things/thermostat/properties at 10:38 AM:

{
    "idx": "thermostat_upstairs",
    "ip": "192.168.1.177",
    "firmware": "1.05b",
    "temperature": 20.5,
    "targetTemperature": 23.5,
    "deviceOn": true,
    "schedulesMode": "off",
    "ecoMode": false,
    "locked": false,
    "systemMode": "heat",
    "fanMode": "medium",
    "mode": "heat",
    "state": "cooling"
}

Thanks a lot :-)

klausahrenberg, 2020-04-03:
It's now also fixed in version 1.02

closed 2020-04-03 by klausahrenberg

fashberg, 2020-04-03:
Hi @JoostHman 

> Update: I saw this has to do with the custom hardware modification, this is not an issue.

Yes, i guess the setting "Relay at GPIO 5" got enabled.
Otherwise this property will not be outputted.

Brewer7, 2023-02-06:
Hi all.
am i got it correct: fan control on fw by Klaus works only with HW mod? i can't get it wok iether on my BAC-002ALW... (ho HW mod yet)